brocaar/chirpstack-application-server #392: how to send data to device with REST API

sslazio2009, 2019-11-30:
### Is this a bug or a feature request?
no
### What did you expect?

### What happened?
Response error   
"error": "authentication failed: jwt parse error: token contains an invalid number of segments",

1.set jwt (copy from configure) in the right top input field
2.POST /api/devices/{device_queue_item.dev_eui}/queue

device_queue_item.dev_eui 3132373772377f04

body 
{
  "deviceQueueItem": {
    "confirmed": true,
    "data": "Vh+9et1/p95c7zJIoL03KRmXtsNsGyM2oEX+H2/xj68=",
    "devEUI": "3132373772377f04",
    "fCnt":  128,
    "fPort": 128,
    "jsonObject": "string"
  }
}
3.Response error

it seem to  i lost step claim.  could you tell how to use it in detail?

### What version are your using?
chirpstack-application-server_3.5.1_linux_amd64.tar.gz
### How can your issue be reproduced?

### Could you share your log output?

sslazio2009, 2019-11-30:
token : 123456789

raphaelruet, 2019-12-05:
Hi sslazio,

To use REST API, the best is to start with the RESTful JSON interface which should be available at `https://<<your_chirpstack_app_server_url>>/api`.

They there are three steps to make it work :
1. Create a valid user using App server's web interface
2. Retrieve a valid JWT using RESTful web interface
3. Use this JWT to execute call to others api routes.

To get a JWT : 
1. Use the `/api/internal/login` route (provide the correct login/password)
2. The JWT is provided in the server's response :

[image]

3. Copy the provided JWT to the right top input field of the page
4. You're ready to go, the RESTful web page will provide your JWT when proceeding requests

Hope this will help you. If not already done, remember you can also use the forum to get help from the community.

closed 2019-12-10 by brocaar