brocaar/chirpstack-application-server #393: API reset counter in activation request

geniv, 2019-11-30:
### Is this a bug or a feature request?
bug: yes

### What did you expect?
I don't expect the counter to reset

### What happened?
If I send payload to activate 

### What version are your using?
api version 1.0.0

### How can your issue be reproduced?
send payloat to url: `POST /api/devices/{device_activation.dev_eui}/activate`

with this body (without counter):
```
{
  "deviceActivation": {
    "appSKey": "string",
    "devAddr": "string",
    "devEUI": "string",
    "fNwkSIntKey": "string",
    "nwkSEncKey": "string",
    "sNwkSIntKey": "string"
  }
}
```
if I have any value on `fCntUp` or `nFCntDown` and I send this payload then value `fCntUp` and  `nFCntDown` are after send in value `0` <- that is totaly wrong.

brocaar, 2019-12-02:
This is not a bug. When you leave fields blank, they fall back to their default values. In this case the default for an `int` is `0`. If you want to maintain the original counter values, you need to include them in your request.

The behavior you are expecting (a partial replace) is usually implemented in the `PATCH` HTTP method: https://developer.mozilla.org/en-US/docs/Web/HTTP/Methods/PATCH.

closed 2019-12-02 by brocaar

geniv, 2019-12-02:
hmm of course, but if I want update only keys? without counters?

Yes, partial replace is for PATH or PUT but in API does not this endpoint variant. 

Thank you for the explanation.