brocaar/chirpstack-application-server #405: First argument must be a string, Buffer, ArrayBuffer, Array, or array-like object

KangGeunSu, 2019-12-26:
TypeError: First argument must be a string, Buffer, ArrayBuffer, Array, or array-like object.
    at index.js:326
    at u (index.js:142)
    at Function.l.from (index.js:155)
    at a.value (JSONTree.js:48)
    at Lo (react-dom.production.min.js:3785)
    at Mo (react-dom.production.min.js:3776)
    at Ro (react-dom.production.min.js:3960)
    at qa (react-dom.production.min.js:5514)
    at Ga (react-dom.production.min.js:5536)
    at Ls (react-dom.production.min.js:5958)
ha @ react-dom.production.min.js:4408

[image]

brocaar, 2019-12-30:
This issue has already been fixed in https://github.com/brocaar/chirpstack-application-server/commit/f84c08c384f56635a7e2930fa936630c51425ebe. The latest test-release contains this fix. Next week I'll release a stable version if no issues pop up.

closed 2019-12-30 by brocaar

KangGeunSu, 2019-12-31:
@brocaar Thank you!