brocaar/chirpstack-application-server #408: Gateway "last seen" should not rely on the gateway time information

mullerch, 2020-01-07:
### Is this a bug or a feature request?

Either a bug or a design choice reconsideration request.

### What did you expect?

In the gateway details page, the "Last seen" field should display the date when the gateway has been last seen.

### What happened?

As the "Last seen" information relies on the date given by the gateway, this information is wrong when the gateway is not at the correct time.

### What version are your using?

ChirpStack Application Server version 3.5.1

### How can your issue be reproduced?

1. Set a wrong date on a gateway
2. Connect to the application server
3. Check the Last seen field

### Could you share your log output?

I don't think this is useful.

Safrone, 2020-02-06:
Agreed in general, both dates might be interesting but last_seen implies real time vs whatever time is on the gw

brocaar, 2020-02-07:
I agree. If one of you would like to create a pull-request, that would be appreciated!

Safrone, 2020-02-07:
Do any of the gateway fields currently correspond to the last_seen or would a new one need to be added? Modified isn't ideal since any api change to the gateway would ostensibly update that.

mmrein, 2020-02-10:
Noticed this issue few days ago, should deffinitely be a server time. 

Also, wouldn't it be better if "lastSeenAt" gets updated if gateway send stats too? Currently it seems to get updated on uplink messages only. 

And third one - there is "updatedAt" timestamp which seems to get updated when gateway send stats. Bug or feature? Btw this timestamp seems to be server time.

brocaar, 2020-02-10:
@Safrone  I think the only actions needed would be

* to replace this line with `time.Now()`:
https://github.com/brocaar/chirpstack-application-server/blob/master/internal/api/as/as.go#L477

* remove the https://github.com/brocaar/chirpstack-application-server/blob/master/internal/api/as/as.go#L463 variable definition.

---

> Also, wouldn't it be better if "lastSeenAt" gets updated if gateway send stats too? Currently it seems to get updated on uplink messages only.

The `lastSeenAt` gets updated on stats, not on uplinks (when `HandleGatewayStats` is called, see the above link).

mmrein, 2020-02-10:
You're right, the gateway time got me once again.

PR #440 (2020-03-16): Fix gateway FirstSeenAt, LastSeenAt and metrics timestamp
... to not rely on gateway time. Not sure about metrics timestamp and how it is used

Fixes #408
commit dfd3b31a (2020-03-16): Use server time instead of gateway reported time in stats (#440)

Closes #408.
closed 2020-03-16 by brocaar