brocaar/chirpstack-application-server #409: InfluxDB 2.0 integration support

Safrone, 2020-01-08:
InfluxDB 2.0 is now in Alpha and the writing API has changed slightly making it incompatible with the current integration.

from the influx 2.0 docs:
https://v2.docs.influxdata.com/v2.0/write-data/#influxdb-api

```
curl -XPOST "http://localhost:9999/api/v2/write?org=YOUR_ORG&bucket=YOUR_BUCKET&precision=s" \
    --header "Authorization: Token YOURAUTHTOKEN" \
    --data-raw "mem,host=host1 used_percent=23.43234543 1556896326" 
```
The biggest change appears to be the auth token header instead of basic auth/no auth from before, should not be too difficult to implement.

There does appear to be planned support in influx 2.0 for 1.0 style data writing according to their blog and theres a workaround using telegraf for now:

https://www.influxdata.com/blog/influxdb-2-migration-path-instrumentation/

proffalken, 2020-01-24:
I came here to ask about this, as this is also the way that you authenticate to their new managed service as well as bespoke servers.

kaciker, 2021-02-05:
Some new for this integration InfluxDB v2 ??

wereii, 2021-04-23:
So, I am playing with that PR of mine - it works as it should, the metrics are being filled in. 
Though, if anyone could point me to some sensible InfluxdbV2/Flux tutorial I would greatly appreciate it.

I am trying to mimick the Channel Usage graph as shown [in the old influxdb integration demonstration by Orne](https://forum.chirpstack.io/t/lora-app-server-influxdb-integration-with-grafana/1318), but I am having hard time "forcing" the query/grafana to use frequency as the value I want to graph against time.

brocaar, 2021-06-01:
This has been implemented by the above commit.

closed 2021-06-01 by brocaar

j0a0vit0r, 2021-06-16:
Hi everyone. Im new here in git and Im trying to use CS in a project of my university. I already have everything working with de CS network server, application server, devices and so on. I'd like to integrate the influxdb on it, but I have the v1. How can I update to version2? I know I have to clone the repository and then checkout this commit, but after that, what am I suppose to do to have the v2 running in my localhost:8080? Tks :) I've installed CS by the debian/ubuntu tutorial

j0a0vit0r, 2021-06-16:
Hi everyone. Im new here in git and Im trying to use CS in a project of my university. I already have everything working with de CS network server, application server, devices and so on. I'd like to integrate the influxdb on it, but I have the v1. How can I update to version2? I know I have to clone the repository and then checkout this commit, but after that, what am I suppose to do to have the v2 running in my localhost:8080? Tks :) 
ps: I've installed CS by the debian/ubuntu tutorial

brocaar, 2021-06-18:
@j0a0vit0r please see https://www.chirpstack.io/application-server/community/source/.

g0lgs, 2021-06-21:
Hi,

I have tried to follow the instructions for building application server from source (in order to get support for influxdb 2.0) on my headless Ubuntu 20.04 but I don't see instructions for the required Go protocol buffer support (at the link provided).

make test fails with:

build github.com/brocaar/chirpstack-application-server/cmd/chirpstack-application-server: cannot load embed: malformed module path "embed": missing dot in first path element