brocaar/chirpstack-application-server #411: Breaking changes: float type in Influxdb integration with existing custom JS parsing with integer values

istina-iot, 2020-01-09:
### Is this a bug or a feature request?
Dangerous bug

### What did you expect?
After updating lora-app-server `3.4` to chirpstack-application-server `3.7` breaking changes in Custom JS parsers and Influxdb integration.
Success writing to Influxdb parsed by existing JS custom codec data has proper types for `integer` values

### What happened?
Unsuccessfull writing attempt to influxdb with existing data `integer` type

### What version are your using?
`3.7`

### How can your issue be reproduced?
Write some data to influxdb lora-app-server `3.4` with custom JS parsing:
```
function Decode(fPort, bytes, vars) {
    return { mustBeInteger: 1 };
}
```
Then try write some data to influxdb chirpstack-application-server `3.7` with existing custom JS parsing

### Could you share your log output?
POST from chirpstack-application-server to influxdb
```
device_frmpayload_data_mustBeInteger,application_name=xxxx,dev_eui=xxxxx,device_name=xxx,f_port=3 value=1.000000
```
Influxdb response
```
partial write: field type conflict: input field "value" on measurement "device_frmpayload_data_mustBeInteger" is type float, already exists as type integer
```

brocaar, 2020-01-13:
This has always been a bit dangerous. Technically, JS only supports floats, not integers. There is no way to cast one to an other in the JS functions. The output by Otto (the JS VM in Go) can be sometimes an integer, sometimes a float. I have not investigated the logic behind this. However, I think this should maybe be changed to floats only in the mapping to InfluxDB (https://github.com/brocaar/chirpstack-application-server/blob/master/internal/integration/influxdb/influxdb.go#L74).

Currently when the first measurement is written as integer, then the following measurements must also be of type integer. The dangerous part is that this typing is not explicit.

proffalken, 2020-01-24:
Yeah, this is an InfluxDB thing in which you can't change the data type once a measurement has been created.

I've gotten around this in the past by creating a dummy entry for each new measurement with a value of `0.0`, but that's going to be a real pain (not to mention a really bad idea!) when this is being done dynamically by Chirpstack.

Completely agree with @brocaar to always send floats if we can...

brocaar, 2020-03-18:
I have looked into this a bit further, I believe implicitly the output from the JS codec is now casted to a float, which could explain the change in behavior mentioned by @istina-iot .

The InfluxDB integration unmarshals the `UplinkEvent` JSON (and thus numeric values into a float), and casts all values accordingly to the InfluxDB data type. As in JSON there is only a `number` type (https://www.w3schools.com/js/js_json_datatypes.asp), I believe the only correct way to cast a measurement is to the InfluxDB float type.

In earlier versions there was an Object field with type `interface{}`, holding the output from the JS codec. I did notice that the Otto VM library used for executing the JS code did sometimes return int types instead of floats, depending the code that was being executed.

When introducing the Protobuf based integration events, the Object field was replaced by a ObjectJson field of type `string` (to be able to map to a Protobuf type).

This change could have caused that what was previously casted as an int, is now (correctly) casted to a float. I'm sorry for the issue that is causing this, but I think this should be the correct behavior.

closed 2020-03-18 by brocaar