brocaar/chirpstack-application-server #412: Request queuing multiple downlink packets close together causes duplicate fCnt numbers

darkfader, 2020-01-13:
Still seeing this exact same thing: https://github.com/brocaar/chirpstack-application-server/issues/245

Additionally, I get "Internal Server Error" when I just restarted a node and trying to queue stuff on it. But perhaps that is by design.

Safrone, 2020-02-13:
I think the restarting issue has to do with needing to get an uplink to verify the device nonce after sending a join accept. Until then you cant be sure the downlink you're queuing will be properly decoded.

darkfader, 2020-03-31:
Even when the node is not restarting I get "device security-context out of sync" sometimes when trying to queue when the link is bad (lots of uplinks, few downlinks). So I guess it's also when frame counter don't match up close enough?

brocaar, 2020-04-03:
:man_facepalming: 

I think this might have been a regression during a cleanup: https://github.com/brocaar/chirpstack-application-server/blob/master/internal/api/external/device_queue.go#L59. `storage.DB()` should have been `tx` to make sure the row is locked during the whole transaction.

commit e6951254 (2020-04-03): Fix passing the database transaction.

Closes #412.
closed 2020-04-03 by brocaar