brocaar/chirpstack-application-server #420: mydevices integration can't be saved

bryanstein, 2020-02-01:
### Is this a bug or a feature request?
Bug
### What did you expect?
A new integration should be added to a new or existing application upon clicking "Create Integration"
### What happened?
I can click "Create" from the integrations menu on an application and choose myDevices and an appropriate endpoint but when I go to save the integration by clicking "Create Integration" I'm prevented from saving it.
### What version are your using?
3.8.0-test.1
### How can your issue be reproduced?
Create a new application or select an existing application:
Select the "INTEGRATIONS" menu item
Select "CREATE"
Select "myDevices.com" from the "Integration Kind" drop down
Select "Cayenne" or any other option from the myDevices endpoint drop down
Select "CREATE INTEGRATION" and it will fail to save
### Could you share your log output?
There is no log output

brocaar, 2020-02-03:
I'm unable to reproduce your issue. I can create and update the myDevices integration without error and data is persisted. 

Could you provide any browser logs that could help debugging this further?

bryanstein, 2020-02-05:
Sorry this seems to have resolved itself after using a different browser. I went from Vivaldi to Chrome and it worked :o/

closed 2020-02-05 by bryanstein