brocaar/chirpstack-application-server #422: ADD objectJSON data to the uplink window(line), under DEVICE DATA tab into Application/app/Devices/dev page

ric2498, 2020-02-06:
### Is this a bug or a feature request? 
feature request

### What did you expect?
Could we add the objectJSON data to the the uplink window(line) ? ; it will eliminate the needs to open the line to see the objectJSON data

### What happened?
n/a
### What version are your using?
3.7.0

### How can your issue be reproduced?
n/a

### Could you share your log output?
n/a

brocaar, 2020-04-03:
I'm not sure if this would look well, as the `objectJSON` could contain multiple (nested) items. Could you share how you think this should look like?

ric2498, 2020-04-03:
Hi,

Thanks Orne,

You will find into this pdf file the way I think it could look.

Richard.

From: Orne Brocaar 
Sent: Friday, April 3, 2020 10:08 AM
To: brocaar/chirpstack-application-server 
Cc: ric2498 ; Author 
Subject: Re: [brocaar/chirpstack-application-server] ADD objectJSON data to the uplink window(line), under DEVICE DATA tab into Application/app/Devices/dev page (#422)

I'm not sure if this would look well, as the objectJSON could contain multiple (nested) items. Could you share how you think this should look like?

sophiekovalevsky, 2020-04-03:
@ric2498 it looks like pdf is missing.

ric2498, 2020-04-03:
sorry you should have it now.

From: Kiara Navarro 
Sent: Friday, April 3, 2020 10:48 AM
To: brocaar/chirpstack-application-server 
Cc: ric2498 ; Mention 
Subject: Re: [brocaar/chirpstack-application-server] ADD objectJSON data to the uplink window(line), under DEVICE DATA tab into Application/app/Devices/dev page (#422)

@ric2498 it looks like pdf is missing.

closed 2020-10-05 by brocaar