brocaar/chirpstack-application-server #426: DNS name does not exist. (code: 14)

sagar-patel-sls, 2020-02-18:
I have downloaded .exe file from below URL
https://www.chirpstack.io/application-server/overview/downloads/

1. started chirpstack-network-server using CMD
2. using CMD I am starting chirpstack-application-server and it started perfectly but when access ui from browser it will throw below error

`error parsing service config: error from DNS TXT record lookup: lookup _grpc_config.localhost: dnsquery: DNS name does not exist. (code: 14)`

there is no logs available in chirpstack-application-server.
[image]

sagar-patel-sls, 2020-02-20:
Hii @brocaar ,

we have solved this `error parsing service config` error 

we have updated the `gRPC` version in `go.mod` file 
~~google.golang.org/grpc v1.26.0~~ 
google.golang.org/grpc v1.27.0

https://github.com/grpc/grpc-go/releases/tag/v1.27.0

Please review this change...

brocaar, 2020-02-20:
Awesome, thanks for testing this :+1:

commit 0159378f (2020-02-20): Update gRPC dependency to latest version.

Closes #426.
closed 2020-02-20 by brocaar