brocaar/chirpstack-application-server #427: Feature proposition: MQTT authentification for Gateways

hoellejal, 2020-02-19:
### Is this a bug or a feature request?
Feature proposition

### What is it?
It's a new method of authentification for the gateways with the addon mosquitto-go-auth. We link a MQTT key with each Gateway at registration (we can auto generate the key from the UI) and save it in the postgresql database.

### How to implement it?
We add a field MqttKey to the gateway definition in the API, and in the database table gateway. We modify the UI to show the MQTT associated with a gateway and type/generate it when we register a gateway. We use mosquitto-go-auth.

brocaar, 2020-03-17:
I'm not sure if ChirpStack should implement such token.

There are many ways to authenticate a gateway and ChirpStack should not limit users in this. E.g. with GCP and Azure you can use client certificates and the token you are proposing would not be useful. Even in case people are using MQTT, there are many users using VerneMQ or RabbitMQ with MQTT ingestion.

This is a really specific solution. I'm looking forward to feedback, from you and others, on how we (if possible) could make this more generic.

Diss31, 2020-03-18:
If someone is interested to see our basic implementation of this feature, you can check it on this [post](https://forum.chirpstack.io/t/mqtt-authentification-for-gateways/7787).

brocaar, 2020-04-03:
> I'm not sure if ChirpStack should implement such token.

Actually, I do see some use-cases where this might be useful, but I think I would prefer a certificate instead of token. See my comment in the above link for more information how I see this could work

closed 2020-06-29 by brocaar

brocaar, 2020-06-29:
The above implements an option to generate client-certificates for gateways. This works in combination with https://github.com/brocaar/chirpstack-network-server/commit/67e6fb212a746eb77ed12e6dc35921fceb8255ec.

As the client-certificate contains the gateway ID as CN, this can be used in combination with Mosquitto (and other MQTT brokers) for authentication and authorization. E.g.

`mosquitto.conf`

```
listener 8883
cafile /etc/mosquitto/ca.pem
certfile /etc/mosquitto/cert.pem
keyfile /etc/mosquitto/cert-key.pem
allow_anonymous false
use_identity_as_username true
acl_file /etc/mosquitto/acl
require_certificate true
```

`/etc/mosquitto/acl`

```
pattern readwrite gateway/%u/#
``` 

Because of `use_identity_as_username true` and the CN being equal to the gateway ID, `%u` will substitute the gateway ID.

It is important that the `ca.pem` in `mosquitto.conf` is the same CA certificate as which is configured in `chirpstack-network-server.toml` (together with the CA key), as the `ca.pem` is used to validate if the gateway client certificate is valid (= signed by the same CA certificate).

commit 52f1d67a (2020-06-29): Implement option to generate gateway client certs.

Closes #427.