brocaar/chirpstack-application-server #428: UI search broken in 3.8.0-8-g6a0a81b

istina-iot, 2020-02-20:
### Is this a bug or a feature request?
Bug or new feature?
### What did you expect?
UI search worked
### What happened?
UI search broken
### What version are your using?
3.8.0-8-g6a0a81b
### How can your issue be reproduced?
install docker and try search
### Could you share your log output?
[image]

closed 2020-02-20 by brocaar

istina-iot, 2020-02-20:
Same bug 3.8.0-9-g0159378

brocaar, 2020-02-20:
Thanks for reviewing this so quick. I know, I'm checking the wrong field. The input should not be empty instead of the target. Fixing this now.

istina-iot, 2020-02-20:
3.8.0-11-g477367e - ok
Thanks!

commit 14bda040 (2020-02-20): Fix matching empty hstore elements.

Closes #428.