brocaar/chirpstack-application-server #433: uplink events only

shepiloff, 2020-03-02:
### Is this a bug or a feature request?
bug
### What did you expect?
uplink/downlink  messages in live lorawan frames
### What happened?
after upgrade to app server version 3.9.0  downlink messages is gone
### What version are your using?
3.9.0
### How can your issue be reproduced?
just go to the Live Lorawan Frames and you will see only uplink messages

brocaar, 2020-03-03:
This depends on your gateway and packet-forwarder version.

> Previously the first downlink scheduling attempt was logged (visible in the ChirpStack Application Server under the Gateway / Device LoRaWAN frames). This has been changed so that the scheduling attempt acknowledged by the gateway is logged. E.g. when RX1 fails, but RX2 succeeds this will log the RX2 attempt.

This means that when your gateway does not send an ack, it won't be visible in the downlink logs. The latest Semtech Packet Forwarder does send this ACK.

Could you share:

* Gateway vendor / type
* Packet Forwarder version?

shepiloff, 2020-03-03:
Could you share:

    Gateway vendor:  LorixOne 256mb
    Packet Forwarder version:  
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 3.1.0
and
*** Poly Packet Forwarder for Lora Gateway ***
Version: 2.1.0

mmrein, 2020-03-06:
Hi, just to let you know I'm willing to test this next week.

I have test server running an app-server 3.8.0 and LorixOne 256MB (Legacy FW 2019-04-17, semtech packet-forwarder) so I can check before and after update. 

I guess I should be watching Live Frames on gateway and possibly also subscribe to all MQTT topics just to see what's going on. Let me know if you would like me to watch for something specific.

mmrein, 2020-03-10:
Ok, I can see downlinks in LoRaWAN frames as ussual. Semtech packet-forwarder on LorixOne does send ack. 

When device sends ConfirmedDataUp:
```
gateway/[GWID]/event/up ...
application/[AppID]/device/[DevEUI]/rx ...
gateway/[GWID]/command/down ...
gateway/[GWID]/event/ack ...
```

**I see no problem on my system.**

Tested with versions:
- GW: 3.7.1, NS: 3.8.0, AS: 3.9.0
- Packet-forwarder version actually unknown for me.

closed 2020-04-03 by brocaar

shepiloff, 2020-04-24:
I double check this issue and found, when  I use semtech  packet forwarder - all Ok? I see the downlink frame in live loraWan frame tab, but when I use poly packet forwarder - no downlinks in live view, uplink frames only.