brocaar/chirpstack-application-server #446: read configuration file error

geogerli, 2020-03-23:
### Is this a bug or a feature request?
bug
### What did you expect?
no configuration file found
### What happened?
process exits
### What version are your using?
3.9.0
### How can your issue be reproduced?
a read configuration file error occurs ,when the 'chirpstack-application-server.toml' file does not exist.the dependent viper v1.4.0 is ok

shepiloff, 2020-03-25:
exactly. without chirpstack-application-server.toml  app server will not start.  what did you expect?

geogerli, 2020-03-26:
[image]

geogerli, 2020-03-26:
If you change the version of viper in the go.mod file to v1.4.0, it will run normally after compiling

brocaar, 2020-03-26:
@geogerli  your issue was a bit difficult for me to understand as I was not sure how to reproduce it. 

However it is clear now with the screenshot. The issue is that viper (at least < 1.4) detects the binary as config file (when you execute `./chirpstack-application-server configfile`) which for obvious reasons fails. However when you execute `./build/chirpstack-application-server configfile` it does work.

Would you like to create a PR to update the viper dependency?

geogerli, 2020-03-27:
What does PR mean? This bug may be viper's own, and the following steps are required to reproduce it.

[image]

brocaar, 2020-04-03:
> What does PR mean? 

Pull Request :-)

PR #461 (2020-04-03): V3.9 bugfix
Fixes #446
commit 75c0fb62 (2020-04-03): Update viper to fix reading configfile error. (#461)

Closes #446
closed 2020-04-03 by brocaar