brocaar/chirpstack-application-server #455: Timestamp in downlinks in live lorawan frames view.

darkfader, 2020-03-31:
### Is this a bug or a feature request?
timestamp feature in live lorawan frames view.

### What did you expect?
To be able to see if a download is being generate and/or delivered to gateway in time.
By checking timestamp in sub-second accuracy. All accuracy we see now is a minute.
The uplink json data does have timestamp, so I expect it for downlink too. Perhaps even two.

### What happened?
I'm trying to troubleshoot missing downlink messages.

### What version are your using?
3.9.0

mmrein, 2020-04-01:
> To be able to see if a download is being generate and/or delivered to gateway in time.

See [v3.8.0 changelog](https://github.com/brocaar/chirpstack-network-server/releases/tag/v3.8.0): 

> ### Downlink LoRaWAN frame logging
> 
> Previously the first downlink scheduling attempt was logged (visible in the ChirpStack Application Server under the Gateway / Device LoRaWAN frames). This has been changed so that the scheduling attempt acknowledged by the gateway is logged. E.g. when RX1 fails, but RX2 succeeds this will log the RX2 attempt.

That basically does what you want. 

This ack means the gateway successfully scheduled the packet for transmission, but does not tell you if or when it physically transmitted it and I doubt that there is such information (considering semtech packet-forwarder).

I once thought that sub-second accuracy would be nice in live frames too, but if you really want to debug timing issues, you will have to do it on the device/node side, which will tell you most.

brocaar, 2020-10-05:
I'm closing this issue. I'm not sure if this would solve your issue. The timestamp would be set to when the NS generated the downlink. If you have a significant network latency between the GW <> NS, then the timestamp would not help you. E.g. it might be less than a second ago, but together with the uplink latency it might be more. Also you have to take the de-duplication into account.

In the release mentioned by @mmrein the downlink scheduling has been re-factored. In the latest releases there is also a `txack` event and downlink errors (related to timing, but also frequency errors etc...) are visible in the web-interface (device data) and published to the integrations.

closed 2020-10-05 by brocaar