brocaar/chirpstack-application-server #460: Duplicate fCnt received via HTTP integration

darkfader, 2020-04-03:
### Is this a bug or a feature request?
bug

### What did you expect?
I expected duplicate fCnt to be filtered or an indication that it's a duplicate.

### What happened?
I got the same data twice via HTTP integration.
duplicate fCnt 104.
second timestamp was 48 seconds later!
first rssi -73 and then -76.

### What version are your using?
3.9.0

### How can your issue be reproduced?
dunno. try something with fast interval like 1 minute interval.

### Could you share your log output?

brocaar, 2020-04-03:
Are you sure you don't have the disable frame-counter option enabled? ChirpStack Network Server does reject frame-counters that did not increment.

darkfader, 2020-04-03:
Checkbox isn't checked, so no. Can't remember enabling it for this node either.

brocaar, 2020-04-03:
> I got the same data twice via HTTP integration.

Could you clarify at which HTTP endpoint your received this? Please note that ChirpStack does send an error event when a re-transmission is detected, which includes the `fCnt`. So that could also explain why you are seeing the same `fCnt` twice.

darkfader, 2020-04-06:
I used ThingsBoard HTTP uplink integration (on uplink messages only). I'm having the node sending an uptime too, so whenever there is a straight horizontal line (and an extra dot) in the telemetry graph, I know it has been processed twice.

closed 2025-05-02 by brocaar