brocaar/chirpstack-application-server #462: Improve issues creation process.

sophiekovalevsky, 2020-04-05:
### Is this a bug or a feature request?
Neither of both. Contribution.

### What did you expect?

When smebody create an issue about the project, currently there are a couple of questions to answer. This is good, however, I think that it can be improved. I have seen some lately a case where somebody made an issue that was more appropiate to be seen in the forum. 

On the other hand, we can categorize issues by either reporting bugs or make contributions, in this sense, we can ask questions that are targered for such case scenarios. Being said this, my approach would be to have a fine-grained issues categories to lead people to the right scenario. 

### What solutions do you propose?

I would be happy to submit a pull request into the several Chirpstack components that will:
- Update ISSUE_TEMPLATES folder
   - Stating off with application server. Example (https://github.com/sophiekovalevsky/chirpstack-application-server/tree/improve-issue-reports)
   - Network Server
   - Gateway Bridge
   - Concentratord
   - Gateway OS

Or any other required that you may think that will require this. 

What are your thoughts?

brocaar, 2020-04-06:
I think this would be really helpful and could potentially save a lot of time :+1:

sophiekovalevsky, 2020-04-06:
That's a good point. I have created a help message when writing the bug report to guide the people in how to provide the log output as you suggested.

About gateways os, if we want to get the image version throughout a command, I suggest start using **image-buildinfo** yocto feature that basically will allow us to add the **DISTRO** and **DISTRO_VERSIOIN**, and **MACHINE** variables into the file /etc/build . We can make it easier even creating a little script that will print the required information in stdout, something: 

```shell
gateway-os version
```
It will output: 

```
-----------------------
Build Configuration:  |
-----------------------
DISTRO = chirpstack-gateway-os
DISTRO_VERSION = 3.3.0-test.5
MACHINE = raspberrypi3
```

If you think that this can be implemented, I can help with that. I tried to compile the image today setting up the variables to confirm that the variables were written, however, I got some issues with it, then it will take me a few days to complete it. 

By now, I think that in overall the feature is ready to get a review.

sophiekovalevsky, 2020-04-09:
@brocaar, 

I made the modifications for the rest of the components so that it's ready for you to review.

sophiekovalevsky, 2020-04-13:
@brocaar, 

Last week I finally make it work my chirpstack-gateway-os recipes :D but actually not being able to push something until now. I was thinking a bit the modifications for the remaining repositories.

**ChirpStack Gateways OS**

- I noticed that you already have **version** info when someone issue *gateway-config*. So, I have added that to generate the environment section. 
- Since different images contain several components of the stack in an isolated fashion, I think it will be more than enough to know the image version only. 
- About the logs output I propose to print out all the messages that comes from syslog. 

**Chirpstack API**

- I was not sure how to deal with the log output. Then, I leave it as an option without any suggestion. Does this make sense to you? 
- About the environment, I place a hint when somebody uses the api with go. Maybe adding hints for the rest of supported languages?

brocaar, 2020-04-14:
That is great @sophiekovalevsky :+1: With regards to the `chirpstack-api`, I have updated the environment command to `git describe`. As the `chirpstack-api/README.md` lists the Compose commands, I believe the git commit should be enough to replicate the issue (as that command will contain the Go / Rust / NPM / ... lock files).

closed 2020-04-14 by brocaar

sophiekovalevsky, 2020-04-14:
@brocaar sounds great to me. Thank you.