brocaar/chirpstack-application-server #463: Error when accessing gateway page if DNS cannot be reached

alightgoesout, 2020-04-06:
### Is this a bug or a feature request?

bug

### What did you expect?

In the application server, the page of a gateway should be displayed correctly.

### What happened?

The page stays empty and an error toast appears with the message : `context deadline exceeded (code: 2)`

### What version are your using?

3.9.0
In 3.7.0 it works fine, in 3.8.0 I encountered the problem in issue #426. I also tried 3.10-test2 and the issue persisted.

### How can your issue be reproduced?

It seems to appear when the DNS server cannot be reached. This can be reproduced either by editing `/etc/resolv.conf` with a bad nameserver or by disconnecting the server from the DNS. I encountered this when using an offline server that was configured to use Google's DNS (8.8.8.8).

Some context : the application server and the network server are running inside separate docker containers in the same docker network, using docker-compose.

### Could you share your log output?

```
time="2020-04-06T13:07:56Z" level=warning msg="creating insecure network-server client" server="loraserver:8000"
time="2020-04-06T13:08:01Z" level=warning msg="ccResolverWrapper: reporting error to cc: dns: A record lookup error: lookup loraserver on 127.0.0.11:53: dial udp 127.0.0.11:53: operation was canceled"
time="2020-04-06T13:08:01Z" level=error msg="finished unary call with code Unknown" ctx_id=2a5da4b2-a477-49c0-ac06-4a4ac6fe8996 error="rpc error: code = Unknown desc = context deadline exceeded" grpc.code=Unknown grpc.method=Get grpc.service=api.GatewayService grpc.start_time="2020-04-06T13:07:56Z" grpc.time_ms=5011.692 peer.address="[::1]:45684" span.kind=server system=grpc
```

geeks-r-us, 2020-04-22:
Same issue here with 3.10.0 

Application server is running in docker containers and is connected to the same network as the other chirpstack containers.  

When I try to add network-server:

> time="2020-04-22T14:38:21Z" level=warning msg="creating insecure network-server client" server="chirpstack-network-server:8000" 
> time="2020-04-22T14:38:26Z" level=warning msg="ccResolverWrapper: reporting error to cc: dns: A record lookup error: lookup chirpstack-network-server on 127.0.0.11:53: dial udp 127.0.0.11:53: operation was canceled"
> time="2020-04-22T14:38:26Z" level=error msg="finished unary call with code Unknown" ctx_id=4aaf4706-6e4c-4a85-8c2d-9901298d182e error="rpc error: code = Unknown desc = context deadline exceeded" grpc.code=Unknown grpc.method=Create grpc.service=api.NetworkServerService grpc.start_time="2020-04-22T14:38:21Z" grpc.time_ms=5002.185 peer.address="127.0.0.1:60180" span.kind=server system=grpc

But I wasn't able to figure out the reason because name resolution works fine if you run a shell in the container and try to ping chirpstack-network-server.

geeks-r-us, 2020-04-27:
Could be related to: https://github.com/grpc/grpc-go/issues/3572

alightgoesout, 2021-02-18:
I have tried the latest version (3.14.0) and it now works properly. Thanks.

closed 2021-02-18 by alightgoesout

pspada, 2022-08-13:
Adding a comment here in case it could help someone experiencing this issue even on more recent 3.x versions when running chirpstack with docker.

Make sure to get the Network Server `hostname:IP` reference right when configuring it in the Application Server web-interface. That is `chirpstack-network-server:8000` as described in the documentation and quick start guide: https://www.chirpstack.io/project/guides/docker-compose/

I had a typo and after fixing it everything worked as expected.

[image]