brocaar/chirpstack-application-server #465: Allow Multiple Integrations of the same type

proffalken, 2020-04-08:
We'd like to be able to send the same data to multiple end-points for some of our devices.

For example, the [RadioBridge Door/Window Sensor](https://radiobridge.com/products/wireless-ip67-door-window-sensor) needs to be configured via their own console, but we also need to send the data via HTTPS to a couple of other places for analysis.

Obviously I could write my own MQTT client that takes the messages from the queue and `POST`s them to the other locations, but if possible I'd prefer to do this within Chirpstack itself.

### Is this a bug or a feature request?

Feature Request

### What did you expect?

To be able to add more than one HTTP Integration

### What happened?

When trying to add more than one HTTP Integration, it fails with `Object already exists (code: 6)` 

### What version are your using?

Latest Stable Release

### How can your issue be reproduced?

Try to create an additional HTTP Integration (or duplicate of any other type of integration)

### Could you share your log output?

N/A

brocaar, 2020-04-16:
Note that you can comma separate the HTTP endpoints :-)

proffalken, 2020-04-16:
I did not know this... 

Thanks!

closed 2020-05-06 by brocaar