brocaar/chirpstack-application-server #469: Skip SSL verification InfluxDB integration

kjetilmjos, 2020-04-20:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
When enabling the InfluxDB integration with self signed certificates I get an error.
I run all services in docker containers.
## What did you expect?
Having a checkbox to enable skip SSL verification when setting up the InfluxDB integration. 
## Steps to reproduce this issue

Steps:
1. Set up InfluxDB with self signed certificate.
2. Enable the InfluxDB integration.
3. 
4.

## Could you share your log output?
Had to remove SSL on the DB so lost the log messages.

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | 3.10.0  |
| Network Server      |  3.9.0 |
| Gateway Bridge      |  3.8.0 |
| Chirpstack API      |         |
| Geolocation         |   3.3.1  |
| Concentratord       |         |

brocaar, 2020-06-29:
I'm not sure if this would be a useful feature. When you skip the TLS certificate validation it means that you do not validate if the endpoint you are trying to connect to is really then endpoint you think you are connecting to. While technically all data between the client and DB is encrypted, a MITM can still identify itself as the DB with its own (self-signed) certificate and decrypt your data.

kjetilmjos, 2020-06-29:
I can see the safety part of it. I encountered the issue where I have some databases configured with a self signed certificate running in a closed network. I guess SSL is really not necessary when running in a closed network but sometimes it's nice to be hide the information going to the database even if you have control on the network.

closed 2025-05-02 by brocaar