brocaar/chirpstack-application-server #470: Include relevant attributes at {Device,Service} profiles view

sophiekovalevsky, 2020-04-21:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

When someone goes to see whether Services or Device profiles, the only information that it's displayed is the name of each profile. 
In addition of this attribute, I propose to add the following:

- Network server name with their respective anchor link 
- Created at or Updated at or both 

## What is the use-case?

It happens that when someone is within a Device profile, there is no right away information that shows to which network server the Device profile belongs, and it may happen that you want to know this information and probably have a shortcut to go to that Network Server detail view.

As regards to the date attributes, I am not sure if this is like a reaaally relevant but I added them in the proposal from the fact that I see the table with few elements to display. Is this is something on purpose within ChirpStack Application Server to maintain as much as possible a pristine skeleton so that people can use it as a blueprint for their own development process?

The same applies for Service profile.

## Implementation description

The image below express what would be visible in the front-end:

[image]

The ChirpStack API already provides the Network Server ID when it lists the {Device, Service} profiles, then, an inner join query can be implemented in the backend to retrieve as well the name associated to the Network Server ID. 

The anchor link can be generated from the Network Server ID. 

## Can you implement this by yourself and make a pull request?
I would be happy to implement this. Nevertheless, some questions will arise through my mind during the development process, specially in the backend since probably you already have some structure in how to approach this type of case scenarios I want to stick with it.

brocaar, 2020-04-29:
I think it is good to at least add the NS name, as sometimes when dealing with multiple NS instances (EU, US), this is confusing.

As with the created at date, I'm not sure. Would you mind when we leave this out for now? The created at / updated at timestamps are stored for most if not all objects, but it would be inconsistent to show them here but not in other tables and adding an extra column might not fit in all tables.

Please let me know how I can help :)

sophiekovalevsky, 2020-04-30:
That's why I wanted your opinion on this and certainly you have open my eyes in this regard. Having the fact that those timestamps won't be available for all the objects, it makes sense then to leave the dates off scope. 

> As with the created at date, I'm not sure. Would you mind when we leave this out for now? The created at / updated at timestamps are stored for most if not all objects, but it would be inconsistent to show them here but not in other tables and adding an extra column might not fit in all tables.
> 

All right then, I am going to contribute on this in the upcoming days ping you if it's required. 

> Please let me know how I can help :)

sophiekovalevsky, 2020-05-03:
Hi @brocaar, here are some questions that rise while I was implementing this feature. 

At Chirpstack Application Server:

- How will you test the swagger implementation? 
In Makefile, I can not overwrite static/swagger/api.swagger.json rule to point to my fork repo.

- How will you test the integration with chirpstack-api? 
Almost same situation then previous. I only changed ChirpStack API version in mod.go, however it was not updated, since I can not point to my own repo.

- I was wondering whether I need to add some tests for this implementation. I tried to find some test go files but I didn't find. This is correct? 

At ChirpStack API:

- I was not able to develop this at https://github.com/brocaar/chirpstack-api/commit/86168697b3b5eaa7fdd7b903d2416bb070e1caf7 with 271e696e4cd4c83e543555633c17ec1e6ab41ad9. Then I had to checkout to https://github.com/brocaar/chirpstack-api/commit/14752ccdfa59de21b5119b6ff86793cd2a5b38ad 
In this regards I re-generate the gRPC and Protobuf definitions without latest changes from https://github.com/brocaar/chirpstack-api/commit/86168697b3b5eaa7fdd7b903d2416bb070e1caf7 does this make sense to you?

I had a bad time trying to make understand git in setting right indendation, spaces... what's the type of format ChirpStack uses in this regard?

brocaar, 2020-05-11:
Hi @sophiekovalevsky sorry for the delay. Last week was packed :)

Please note that I did release a `chirpstack-api` update with your changes last week, so this should be as simple as updating to the latest version :) You are right that testing the swagger api is a bit painful.

Most of the times when I know the exact impact on the `chirpstack-api`, I make these changes first and tag a new release. After updating the `chirpstack-api` dependency I then can do a `make clean build`.

In some cases, I point my `go.mod` to my local clone (with modifications) and within `static/swagger` I execute `cp ../../../chirpstack-api/swagger/as/external/api/*.json .`

sophiekovalevsky, 2020-05-11:
Hi @brocaar, I hope you are doing well. No problem, I was thinking that you were kind of busy. 

I did see the merge, but I was waiting for your reply on this issue. In the next upcoming days, I will test my changes with latest chirpstack api version and double check if everything works as supposed to be. 

I will get back to you until then.

sophiekovalevsky, 2020-05-12:
@brocaar, 

PR is ready for you to test. :)

commit 150084ad (2020-05-15): Include Network Server Name column in Device and Service Profile List. (#475)

Closes #470.
closed 2020-05-15 by brocaar