brocaar/chirpstack-application-server #471: Include hover at list table views.

sophiekovalevsky, 2020-04-21:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

Whenever someone passes the mouse pointer over a row table at any given list view, fire up an hover change in color for that particular row.  

## What is the use-case?

ChirpStack Application Server allows to play with how many entries will be seen per row. When the ratio rows per page increases there will be more elements in one single view. I think a good asset to improve user experience would be to highlight the row entry to which the user wants to inspect.

Another use case can be ilustrated e.g. if you want to cross Device Name with it's respective battery value. You will need to guide your eye almost to end of the row, which can be difficult to follow if there is no tool to help you to highlight it. 

## Implementation description

The following gif represents how the user will feel this at front-end level.

[image]

The front-end changes would be a React Js component, adding **hover** to each TableRow instance. 

## Can you implement this by yourself and make a pull request?
Yes, I do.

brocaar, 2020-04-29:
Sounds good to me! :-) :+1: 

I guess that would be using this option?

[image]

sophiekovalevsky, 2020-04-29:
Yeah, that's right.  I will prepare changes this week :)

sophiekovalevsky, 2020-05-02:
- Row tables that are not listing items with row per page such as Details at DeviceDetails were not taken into consideration in this feature.

PR #474 (2020-05-05): Add highlight row when user hover it at ui tables
This commit enhnace user table interaction when viewing list items in ui.

Closes #471
commit 56af35c3 (2020-05-05): Add highlight row when user hovers it at ui tables (#474)

Closes #471.
closed 2020-05-05 by brocaar