brocaar/chirpstack-application-server #476: Unable to set device keys when registering new device through grpc

eivindlm, 2020-05-07:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

I ran the following python code to register a new device through grpc:

    from chirpstack_api.as_pb.external import api

    # [ Omitted code for server login ]

    application_id=5  # ID of application on the app server
    device_profile_id="device profile id"
    dev_eui="valid dev eui of a device"
    app_key="valid app key for the device"

    device = api.Device(application_id=application_id,
        description="Some description",
        device_profile_id=device_profile_id,
        name=dev_name,
        dev_eui=dev_eui)

    device_keys = api.DeviceKeys(dev_eui=dev_eui, nwk_key=app_key)

    try:
        response = dsStub.Create(request=api.CreateDeviceRequest(device=device),
            metadata=metadata)
        print(f"Created {dev_name}")
    except grpc._channel._InactiveRpcError as e:
        print(f"Failed to create device {dev_name}")

    try:
        response = dsStub.UpdateKeys(api.UpdateDeviceKeysRequest(device_keys=device_keys),
            metadata=metadata)
    except grpc._channel._InactiveRpcError as e:
        print(f"Failed to set device key for {dev_name}")
        print(e)

## What did you expect?
I expect a device to be created on the referenced application, and the device keys to be set for the newly created device.

## Result
The device is successfully created, but setting the device key fails. It may look like the device needs to be activated before keys can be set through the api, but this is a deadlock, since the device cannot be activated before the api_key is set.

After the device is created by my script, and I have manually entered the device keys in thew web ui, and the device has joined with the server, then the above code runs successfully and is able to set/update the device_keys correctly.

This sounds somewhat similar to #359, but a very far fetch from me not knowing the internals.

## Steps to reproduce this issue

Steps:

1. Fetch an application_id and a device_profile_id from the app server, and insert in the above code
2. Insert a valid dev_eui and app_key for a device in the above code
3. Run the code and observe that device creation is successful, but setting the app_key fails
4. Insert the app_key manually in the web ui
5. Wait until the device joins the network
6. Run the above code again, and observe that setting the app_key is now successful

## Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.10.0  |
| Network Server      |  v3.9.0    |

brocaar, 2020-05-08:
> but setting the device key fails.

Could you share the error of the failure? :) I have not yet tried to validate this, but please note that the web-interface uses the same gRPC API (through a JSON proxy).

eivindlm, 2020-05-08:
Here is the error message, which occurs in the call to `dsStub.UpdateKeys`:

    <_InactiveRpcError of RPC that terminated with:
            status = StatusCode.NOT_FOUND
            details = "object does not exist"
            debug_error_string = "{"created":"@1588922453.292925871","description":"Error received from peer ipv6:[::1]:8080","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"object does not exist","grpc_status":5}"

The device is successfully created from the preceeding call to `dsStub.Create`, and appears on the list of devices for the application. But naturally with "n/a" in the "last seen" column. The exact same python code completes successfully if I first manually enter the app_key in the web interface for the newly created device, and then run the python code again. So actually step 5 is not required in my "steps to reproduce" in the initial report. Simply entering the app_key manually once in the web interface (step 4) allows me to to reset it later from python (step 6).

sonirohit2, 2020-05-17:
I am also not able to add device from /api page.
Command : curl -X PUT --header 'Content-Type: application/json' --header 'Accept: application/json' --header 'Grpc-Metadata-Authorization: Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhdWQiOiJjaGlycHN0YWNrLWFwcGxpY2F0aW9uLXNlcnZlciIsImV4cCI6MTU4OTgxMTk3MiwiaXNzIjoiY2hpcnBzdGFjay1hcHBsaWNhdGlvbi1zZXJ2ZXIiLCJuYmYiOjE1ODk3MjU1NzIsInN1YiI6InVzZXIiLCJ1c2VybmFtZSI6ImFkbWluIn0.xwXtghYtG8E91Spe-xGeLvqZwPduj7aNOjrJCzwCFmc' -d '{ \ 
   "deviceKeys": { \ 
     "appKey": "02a84fee91004daf326afb30d938d14b", \ 
     "devEUI": "3cc1f605000243ca", \ 
     "genAppKey": "00000000000000000000000000000000", \ 
     "nwkKey": "00000000000000000000000000000000" \ 
   } \ 
 }' 'http://192.168.0.38:8080/api/devices/3cc1f605000243ca/keys'

Response body : 
{
  "error": "object does not exist",
  "code": 5,
  "message": "object does not exist",
  "details": []
}
Device is already added in UI
[image]

brocaar, 2021-06-15:
You have to call `CreateKeys` before you can call `UpdateKeys` :)

closed 2021-06-15 by brocaar