brocaar/chirpstack-application-server #479: Theme Gateway Details Table same as Device Details.

sophiekovalevsky, 2020-05-17:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

At Gateway Details level there is a Table that contains GatewayID, Altitude, GPS coordinates and Last seen at attributes. The main point of this is to make it similar as Device Details. 

## What is the use-case?

If there is a special treament to show the aformentioned information displayed differently than the rest? 

If not, then I think we can take advantage of the same format that there is already in Device Details. The use case would be to have the same format applied to similar information in all over the place. 

## Implementation description

I would take the same approach as it's in Device Details. In a nutshell, we will go from here:

[image]

to:

[image]

## Can you implement this by yourself and make a pull request?
I would be happy to implement it.

mmrein, 2020-05-18:
Although the original style may look better on smaller screens, I would prefer to have same format for both.

One thing I would like to mention here is a small bug I found recently on gateway details, but it is on device details page as well. "Last seen at" information (and some others) is loaded on page refresh, but not when you change or click tabs (details / configuration / frames ...). Making it refresh on such actions could improve user experience, I believe.

It is probably separate issue and I did not check any details about it, but since you will be rebuilding the page I thouhgt it would worth mentioning.

brocaar, 2020-05-19:
@sophiekovalevsky yes, I agree :+1: 

@mmrein I think this could apply to a few other pages too. The object is loaded higher in the object tree. This means that changing tabs does not create a new API request. This makes changing tabs faster (e.g. no reload on tab change), but could be confusing when you expect to load a new page.

Lets separate this issue. Maybe a better option could be a refresh button to make this more explicit?

sophiekovalevsky, 2020-05-19:
@brocaar I will move forward into this :) 

> @mmrein I think this could apply to a few other pages too. The object is loaded higher in the object tree. This means that changing tabs does not create a new API request. This makes changing tabs faster (e.g. no reload on tab change), but could be confusing when you expect to load a new page.

I agree in this regard. I prefer to have a button to explicit reload that portion of the content rather than requesting extra calls. I can collaborate on this as well. @mmrein would be okey to you to create a new issue about what we have discussed?

mmrein, 2020-05-20:
@brocaar, @sophiekovalevsky 
Agreed, I'll write that up.

commit 224c47af (2020-05-21): Theme Gateway Details Table (#480)

This enhancement refactor Gateway Details Table
to meet same unified structure over all tables details in ui.

Closes #479.
closed 2020-05-21 by brocaar