brocaar/chirpstack-application-server #484: Implement a status in internal service

maxreb, 2020-06-01:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

I would love to see the if the jwt token I use is valid. When I login via `internal/login` I get an immediate result if the user/pass is correct. But when I use the API-Key I don't know if its correct or if its revoked. 

## Current Workarround

I create a new `UserServiceClient` and ask for a list of 10 pplz, if it succeed I know I am authenticated.

## Implementation description
e.g. `login/status` returns: 
```
message StatusResponse{
   bool is_authenticated = 1;//if the jwt is valid and you are authenticated
   bool is_authorized = 2;//if the jwt is valid and you are authorized as an user
   bool is_active = 3;//if the user is currently marked as active
   bool is_admin = 4;//if the user/api key has admin rights
}
```

## Can you implement this by yourself and make a pull request?
I could try if you are interested

brocaar, 2020-06-02:
> when I use the API-Key I don't know if its correct or if its revoked.

I believe you already do know this, by looking at the returned status code? E.g. when the token is invalid, revoked or expired, you get an authentication related code + message.

Please note that an `internal/login` returned token is for ChirpStack internal usage. These endpoints might change any time and are not intended for integration.

There is already an `internal/profile` endpoint which might return some of the information you are looking for, but again, the output is intended for internal usage within the ChirpStack project (e.g. web-interface) and can change any time.

maxreb, 2020-06-03:
Thanks for your fast response! 
I wrote an UI which uses the same login credentials as the chirpstack api server. This way I hadn't to implement my own authentication/authorization system. But if you say its for internal only and shouldn't be used for integration I will now code my own authentication and use an API-Key to control the chirstpack's api server.

closed 2020-06-03 by maxreb