brocaar/chirpstack-application-server #490: Support multiple server address for redis cluster

JohnRoesler, 2020-06-13:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

## What is the use-case?

The underlying redis library accepts a `[]string` for server addresses, while the current config only allows for one address. For a redis cluster connection to be HA, the configuration needs to accept multiple addresses so that in the event of a cluster failover due to a node going down, the connection can use a different redis node.

## Implementation description

```
Redis settings
#
# Please note that Redis 2.6.0+ is required.
[redis]
# Redis server or servers...
servers=[
    “10.10.10.10”, “10.10.10.11”
]

username=“foo”

password=“bar”

database=0

# Redis Cluster.
#
# Set this to true when the provided URL is pointing to a Redis Cluster 
# instance.
cluster=false

# The master name.
#
# Set the master name when the provided URL is pointing to a Redis Sentinel
# instance.
master_name=""

# Connection pool size.
#
# Default is 10 connections per every CPU.
pool_size=0
```

Should this still accept / support the prior `url=` setting and optionally parse that if it's provided for a non-cluster config? Or should that setting go away in favor of these new settings?

## Can you implement this by yourself and make a pull request?
yes

JohnRoesler, 2020-06-13:
@brocaar how does this implementation look? https://github.com/brocaar/chirpstack-application-server/compare/master...JohnRoesler:redis-servers?expand=1 wondering if the `url` option should stay or be deprecated.

mmrein, 2020-06-15:
@JohnRoesler the url option reminds me when multiple-servers option was agdded to gateway bridge: https://github.com/brocaar/chirpstack-gateway-bridge/pull/14, so it should work with existing configurations too.

JohnRoesler, 2020-06-15:
@mmrein that's what I was thinking as well - either URL - or - servers + password + database, should be set.

brocaar, 2020-07-01:
Hi @JohnRoesler sorry for the delay. I agree that the `url` option should continue to work in backwards compatible way.

How this could be supported is:

* If `url` is detected, parse it and set the `server:port` to servers, set the password and database
* In the `configfile.go` configuration template, remove the `url` in favor of `servers`

This means that:

* When the "deprecated" `url` is set, it will have priority over what is set in `servers`
* Somebody who is using the `configfile` sub-command will automatically be migrated to the new-style configuration :)

What do you think?

JohnRoesler, 2020-07-01:
@brocaar I think that makes sense as you've described.

JohnRoesler, 2020-07-01:
@brocaar i updated the pull request #494 - how is that looking?

closed 2020-07-16 by JohnRoesler

bitchkat, 2020-08-04:
Does this configuration work if redis and sentinel require password authentication?  Currently, I have to set the password in the url and this works connecting to sentinel and discovering the redis master.  However, it then connects to redis completely unauthenticated which fails  In my case, redis and sentinel have the same password but that should not be a requirement.

JohnRoesler, 2020-08-05:
@bitchkat it looks like only the sentinel password is being set: https://github.com/JohnRoesler/chirpstack-application-server/blob/master/internal/storage/storage.go#L75

The config for chirpstack would need to be updated to accept an additional password, one for redis and one for sentinel. They're both available on the failover client https://pkg.go.dev/github.com/go-redis/redis/v8?tab=doc#FailoverOptions