brocaar/chirpstack-application-server #491: Add sticky header on tables

sophiekovalevsky, 2020-06-15:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

When someone is navigating over a table, it may happen that the records can reach up to 500 per page. If this happen, as the user scrolls down it will lost visibility of header cells. I propose to take advantage of sticky header at tables so that users can scroll down and still maintain header visibility. 

## What is the use-case?

The use case will apply for all existing tables. The most common use case would be when someone is listing Devices. 

## Implementation description

[image]

Material UI already provides tools to enable this using adding [`stickyHeader aria-label="sticky table"`](https://material-ui.com/api/table/) to `Table` component and adding [`TableHead`](https://material-ui.com/api/table-head/). 

## Can you implement this by yourself and make a pull request?
Yes, I can.

brocaar, 2020-06-26:
Would the height of the table automatically match the remaining height of the screen? It looks like in your example you first scroll down to see the bottom of the table and then you scroll through the table.

sophiekovalevsky, 2020-06-29:
@brocaar, right now the height does not adjust to the high of the screen, a max. height is being used instead.

The effect that happens on the animation is because the height of the main page exceeds the height of my screen. What I can try to dois  match the height of the table with the height of the current screen. Does this make sense to you?

brocaar, 2020-07-01:
Thanks @sophiekovalevsky. to be honest, I'm not sure if I would be in favor of this feature. This would require a couple of "tricks" to make this work properly and I'm not sure if it will be worth the effort.

sophiekovalevsky, 2020-07-01:
@brocaar, all right. Thanks for letting me know.

closed 2020-07-01 by sophiekovalevsky