brocaar/chirpstack-application-server #492: Add loader on tables while fetching data

sophiekovalevsky, 2020-06-15:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

When someone is listing records that may take a while (few seconds), would be great to show a representative loader icon while waiting. 

## What is the use-case?

The use case will apply for all existing tables that fetches records from ChirpStack API. One use case would be when someone is listing a lot of devices, and the following evens are triggered:

 - Rows per page
 - Change page

## Implementation description

I was thiking in having the ChirpStack icon with some flashing effect that provides a sense that is waiting for records to come (flow of flashing from up to bottom). I have done something pretty barebone to illustrate my idea. I know that ChirpStack logo should not be modified in first instance, however, it was required to explain what I was trying to do.  
 
[image]

If this is something that would be accepted, I will require to have the ChirpStack icon in SVG format so that I can make the effects on strokes as supposed to be. 

Aside of this, the implementation can be approach as following:

- Have a new component where the SVG path will be
- Create a variable within state at `DataTable` that will be modified any time at the beginning of the event: e.g. when user triggers any of the aformentioned events and when api completes the fetch from ChirpStack API.  

## Can you implement this by yourself and make a pull request?
Yes, I can.

brocaar, 2020-06-26:
I do like to show a kind of loader animation, but I think it would be better to make this a global component. E.g. some websites show a loader bar at the top of the screen. An example is https://material-ui.com, you'll see it when you open a different page.

If this would be hooked up to the store classes, then this could automatically show an animation when there are requests pending.

sophiekovalevsky, 2020-06-29:
@brocaar it took me a while to figure out where it was that line. Let's me see how can I make that feature global,

sophiekovalevsky, 2020-07-06:
@brocaar, FYI, I haven't a time last week to start working on this and probably this week I won't as well. Then, bear in bind that this make take me. In anyway, once I am about to start doing something on this, I will ping you. Thanks in advance,

brocaar, 2020-07-13:
@sophiekovalevsky no problem :-) As an example, you could take a look at the notification messages. E.g. in case of an create / update / delete or error notification, each store class emits an event with a message. The The `Notifications` object subscribes to events and shows the current notifications:

https://github.com/brocaar/chirpstack-application-server/blob/master/ui/src/components/Notifications.js#L62

I would imagine a similar approach for a loader. Each each store will trigger a loader increment before making a request and a decrement on response (either success or error). When the counter is positive, the loader component is displayed, else it is hidden :-) 

Maybe some of these components can be useful: https://material-ui.com/components/progress/

sophiekovalevsky, 2020-07-15:
@brocaar sounds good to me, let me see how it goes from my side and once is good I will let you know.

sophiekovalevsky, 2020-07-17:
@brocaar, 

I was thinking about where to set the loader and what type of color. So far we have like two primary colors on UI and as I want to set the loader just right below top bar this is how it will look like: 

[image]

Maybe my impressions is wrong but does secondary color make sense? I see it just too much overwhelming compared to blue one? What do you think?

sophiekovalevsky, 2020-07-17:
Here is another proposal:

Move the loader to the top and use some kind of variant from primary color. 

[image]

sophiekovalevsky, 2020-07-22:
@brocaar, I have finished the implementation of this enhancement. A couple of thoughts: 

Does this implementation should be available for all api calls rather than only for create, update and delete actions? 

There are some api calls that don't take a long time, I think in this regard, probably I should show the loader only when a minimum time has been elapsed. You can test this and let me know your thoughts ? 

I will push the PR ready for review. 

Thanks,

brocaar, 2020-07-28:
Thanks @sophiekovalevsky :+1: I will test this in the coming days :)

closed 2020-08-11 by sophiekovalevsky