brocaar/chirpstack-application-server #495: Organizations is missing from side navigation dropdown when more than 10 organizations are created.

micah-vence, 2020-06-26:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
After creating my 11th active organization I was not able to select my first organization from the side navigation dropdown

## What did you expect?
I expect to all active organizations to be accessible from the side navigation dropdown.

## Steps to reproduce this issue

1.  Create 11 Organizations.
2.  Verify all 11 organizations appear in the side navigation dropdown.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.8.0  |

brocaar, 2020-06-29:
It is not intended to show all organizations (e.g. when you have 1000 organizations, this would be a very large dropdown). Instead it shows the first 10 and you can search within the field (it will auto-complete with the 10 closest matches).

micah-vence, 2020-06-29:
Hi Brocaar,
Thank you for the response, I completely missed the fact that the dropdown was searchable.  I am able to search for all my organizations and therefore closing this ticket.
Thanks again.

closed 2020-06-29 by micah-vence