brocaar/chirpstack-application-server #497: Failed Build with docker

symovs, 2020-06-28:
Hi

I tried to build the source code with docker using the command given in documentation 

docker-compose run --rm chirpstack-application-server bash

But it failed with the following error. Please help. I am using the latest cloned version of the application source file from Github. 

[image]

- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

## What did you expect?

## Steps to reproduce this issue

Steps:

1.
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

symovs, 2020-06-28:
I am trying to build on a Raspberry PI zero W with ARM V6 processor.

brocaar, 2020-06-29:
Currently building the Docker image on ARM is not supported.

closed 2020-06-29 by brocaar