brocaar/chirpstack-application-server #498: Last seen not updated with BasicStation backend

mullerch, 2020-06-29:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

Display the gateway list, with gateways connected through the ChirpStack Gateway Bridge with the Basic Station backend.

The "Last seen" column for the gateways with BasicStation is not correct. A connection with BasicStation backend never refreshes the "Last seen" date. If the gateway was never connected it shows "Never". If it has been connected through UDP backend before it shows the last connection time with this backend.

## What did you expect?

The Last seen column is correct.

## Steps to reproduce this issue

Steps:

1. Setup server with CSGB (BasicStation backend)
2.  Setup GW (with BasicStation)
3. Connect the GW to the server
4. Register GW in the server and show Gateways list

## Could you share your log output?

Not sure this is relevant as it shows nothing strange.

## Others

BasicStation backend does not send stats. If Last seen is updated on stats reception, it will not get updated.

If that's the case, best would be that Last seen get's also updated with an uplink reception. At least it should detect gateways with BasicStation backend to not display Last seen as it is false and therefore misleading.

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.10.0  |
| Network Server      | v3.9.0 |
| Gateway Bridge      | v3.8.0 |
| Chirpstack API      |  - |
| Geolocation         |   - |
| Concentratord       |   -  |

andryyy, 2020-06-29:
I have found that only GW stats update the last seen flag. 

The "up" I see in MQTT as well as the devices that sent data over that GW did not update the last seen stamp.

So I guess it is bound to the last stat package and therefore, as you see in the docs, not (yet?) supported with BS.

mullerch, 2020-06-29:
That was my hypothesis too. I did not dig in the code to validate it as for now it's not a high priority to me.

brocaar, 2020-10-01:
I believe this has been fixed already :)

closed 2020-10-01 by brocaar