brocaar/chirpstack-application-server #500: Raspberry Pi Zero - ARM V6 - Compiling From Source

symovs, 2020-06-29:
Hi 

I went through https://www.chirpstack.io/application-server/community/source/. I am confused about how to "Install the C++ implementation of protocol buffers and Go support by following the GO support for Protocol Buffers installation instructions." The page says "It has been superseded by the google.golang.org/protobuf module." Which one should I install ? Ideally, a set of commands to install this would be great. Thanks in advance for any help. 

Best Regards
Syam

- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

## What did you expect?

## Steps to reproduce this issue

Steps:

1.
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

sophiekovalevsky, 2020-06-29:
@symovs I think this is not a bug report related to ChirpStack Application Server but to protocol buffers setup, then, I suggest to look into their official documentation. Even though, you may probable want to take a closer look at https://github.com/brocaar/chirpstack-api where and see how the protocol buffers are being generated to other sort of languages.

brocaar, 2020-07-13:
@symovs I suggest to take the Docker Compose approach. This will (currently) not work on a Raspberry Pi (ARM), but `make build` and `make snapshot` within Docker Compose will crosscompile for ARM so that you can copy the binaries to and ARM based device.

closed 2020-07-13 by brocaar