brocaar/chirpstack-application-server #503: Disable frame-counter validation does not seem to work

andryyy, 2020-07-04:
- [X] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
I checked "Disable frame-counter validation", but the counter validation still seems to be active. No data is pushed to my HTTP integration.
The network server logs `error="get device-session error: frame-counter did not increment"`, in the device logs I see similar errors about non-increasing frames.

## What did you expect?
I expect it to log the errors but still send the device data to the application integration.

## Steps to reproduce this issue
I seem to have a buggy TBDW100 sensors that sometimes (fast door open/close intervals) seems to "forget" increasing the frame counter. That's a nasty bug for a door sensor, but for now disabling the counter validation seems to be my only hope...

## Could you share your log output?

Here you see the first successful uplink arriving just fine, while the following uplinks did not increment. 

```shell
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=fcc23dfffe0b700a stats_id=dfebc745-5545-456c-b52e-28f4b950f62a
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="gateway updated" ctx_id=dfebc745-5545-456c-b52e-28f4b950f62a gateway_id=fcc23dfffe0b700a
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="finished client unary call" ctx_id=dfebc745-5545-456c-b52e-28f4b950f62a grpc.code=OK grpc.ctx_id=b2e1541f-7596-4428-a3b2-bd1c4bdfe9c9 grpc.duration=4.790367ms grpc.method=HandleGatewayStats grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=fcc23dfffe0b700a uplink_id=e9fcf7a8-ffa7-48ef-9442-065c9af3af20
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="uplink: frame(s) collected" ctx_id=2f40401d-b965-4506-abd2-97005c59e522 mtype=UnconfirmedDataUp uplink_ids="[e9fcf7a8-ffa7-48ef-9442-065c9af3af20]"
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="sent uplink meta-data to network-controller" ctx_id=2f40401d-b965-4506-abd2-97005c59e522 dev_eui=58a0cb0000109687
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="device gateway rx-info meta-data saved" ctx_id=2f40401d-b965-4506-abd2-97005c59e522 dev_eui=58a0cb0000109687
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="device-session saved" ctx_id=2f40401d-b965-4506-abd2-97005c59e522 dev_addr=01a3b01f dev_eui=58a0cb0000109687
chirpstack-network-server_1      | time="2020-07-04T10:41:53+02:00" level=info msg="finished client unary call" ctx_id=2f40401d-b965-4506-abd2-97005c59e522 grpc.code=OK grpc.ctx_id=e61f8de7-3383-4a75-8675-cb29af773301 grpc.duration=7.385256ms grpc.method=HandleUplinkData grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:41:59+02:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=fcc23dfffe0b700a uplink_id=b0f833d0-060f-46d7-bc16-573f3ea04a54
chirpstack-network-server_1      | time="2020-07-04T10:41:59+02:00" level=info msg="uplink: frame(s) collected" ctx_id=7d206b12-f1a4-49db-9a5d-6215f5f336dd mtype=UnconfirmedDataUp uplink_ids="[b0f833d0-060f-46d7-bc16-573f3ea04a54]"
chirpstack-network-server_1      | time="2020-07-04T10:41:59+02:00" level=info msg="finished client unary call" ctx_id=7d206b12-f1a4-49db-9a5d-6215f5f336dd grpc.code=OK grpc.ctx_id=df3f44ca-ace2-4d1f-86ea-5e354cf490c5 grpc.duration=2.828856ms grpc.method=HandleError grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:41:59+02:00" level=error msg="uplink: processing uplink frame error" ctx_id=7d206b12-f1a4-49db-9a5d-6215f5f336dd error="get device-session error: frame-counter did not increment"
chirpstack-network-server_1      | time="2020-07-04T10:42:06+02:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=fcc23dfffe0b700a uplink_id=79f33e23-afa9-4b76-8a12-f2f3e2d2f4f5
chirpstack-network-server_1      | time="2020-07-04T10:42:06+02:00" level=info msg="uplink: frame(s) collected" ctx_id=b3d16134-e4a6-4917-9592-0e5cbea696d0 mtype=UnconfirmedDataUp uplink_ids="[79f33e23-afa9-4b76-8a12-f2f3e2d2f4f5]"
chirpstack-network-server_1      | time="2020-07-04T10:42:06+02:00" level=info msg="finished client unary call" ctx_id=b3d16134-e4a6-4917-9592-0e5cbea696d0 grpc.code=OK grpc.ctx_id=7834a27b-d39a-4841-ac97-d697696ad184 grpc.duration=3.684425ms grpc.method=HandleError grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:42:06+02:00" level=error msg="uplink: processing uplink frame error" ctx_id=b3d16134-e4a6-4917-9592-0e5cbea696d0 error="get device-session error: frame-counter did not increment"
chirpstack-network-server_1      | time="2020-07-04T10:42:13+02:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=fcc23dfffe0b700a uplink_id=b5525d31-7ca6-4eab-af95-d83dde291bbc
chirpstack-network-server_1      | time="2020-07-04T10:42:13+02:00" level=info msg="uplink: frame(s) collected" ctx_id=1a6e8e1b-1a12-4468-ba8b-8b94fc1d010a mtype=UnconfirmedDataUp uplink_ids="[b5525d31-7ca6-4eab-af95-d83dde291bbc]"
chirpstack-network-server_1      | time="2020-07-04T10:42:13+02:00" level=info msg="finished client unary call" ctx_id=1a6e8e1b-1a12-4468-ba8b-8b94fc1d010a grpc.code=OK grpc.ctx_id=97396185-2773-40c6-b3ad-03669779ed82 grpc.duration=3.334825ms grpc.method=HandleError grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:42:13+02:00" level=error msg="uplink: processing uplink frame error" ctx_id=1a6e8e1b-1a12-4468-ba8b-8b94fc1d010a error="get device-session error: frame-counter did not increment"
chirpstack-network-server_1      | time="2020-07-04T10:42:19+02:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=fcc23dfffe0b700a uplink_id=b0a8ade8-5933-49d4-bb31-90b91866f382
chirpstack-network-server_1      | time="2020-07-04T10:42:19+02:00" level=info msg="uplink: frame(s) collected" ctx_id=91d014a3-6d1f-4a57-ac71-78acc6a99603 mtype=UnconfirmedDataUp uplink_ids="[b0a8ade8-5933-49d4-bb31-90b91866f382]"
chirpstack-network-server_1      | time="2020-07-04T10:42:19+02:00" level=info msg="finished client unary call" ctx_id=91d014a3-6d1f-4a57-ac71-78acc6a99603 grpc.code=OK grpc.ctx_id=0923c124-a9df-409b-9fa3-bb3db18c1c0e grpc.duration=3.488795ms grpc.method=HandleError grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:42:19+02:00" level=error msg="uplink: processing uplink frame error" ctx_id=91d014a3-6d1f-4a57-ac71-78acc6a99603 error="get device-session error: frame-counter did not increment"
chirpstack-network-server_1      | time="2020-07-04T10:42:23+02:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=fcc23dfffe0b700a stats_id=f772aaba-0621-42bd-9d50-0628e0b7a808
chirpstack-network-server_1      | time="2020-07-04T10:42:23+02:00" level=info msg="gateway updated" ctx_id=f772aaba-0621-42bd-9d50-0628e0b7a808 gateway_id=fcc23dfffe0b700a
chirpstack-network-server_1      | time="2020-07-04T10:42:23+02:00" level=info msg="finished client unary call" ctx_id=f772aaba-0621-42bd-9d50-0628e0b7a808 grpc.code=OK grpc.ctx_id=5a40739d-d148-4415-8b08-a9f6f1d2a6da grpc.duration=4.46873ms grpc.method=HandleGatewayStats grpc.service=as.ApplicationServerService span.kind=client system=grpc
chirpstack-network-server_1      | time="2020-07-04T10:42:25+02:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=fcc23dfffe0b700a uplink_id=964bb5d4-bac5-4d2d-9733-e9eca6e66ffa
chirpstack-network-server_1      | time="2020-07-04T10:42:26+02:00" level=info msg="uplink: frame(s) collected" ctx_id=243d7a03-d111-43cb-8f42-f2201a5f5163 mtype=UnconfirmedDataUp uplink_ids="[964bb5d4-bac5-4d2d-9733-e9eca6e66ffa]"
[... cut ...]
Here you see only one uplink (the first, frame-counter valid) point in Grafana:

[image]

I sadly did not save the "LoRaWAN frames" output where it said "error" a few times. But I remember it just logged "frame counter did not increase", not much more.

## Your Environment

I am running the latest ChirpStack components. GW is a Lorix One. I just flashed the ChirpStack OS to an SD card, the problem remained. It did exist with "LoRiX OS". Just saying for the sake of completeness. I know the core of the problem is the broken device... :)

andryyy, 2020-07-04:
I just waited for it to happen again, now I can show you the output in the web app, too:

The collapsed errors are exactly the same as the opened error box.

[image]

mmrein, 2020-07-09:
Have youu tried re-activating the device? Some device setting changes does require that step. 

You can also try to set 0 to Uplink frame-counter in the same tab (requires re-activation anyways).

andryyy, 2020-07-09:
Thanks, will try with a re-activation. :)

closed 2020-08-31 by brocaar