brocaar/chirpstack-application-server #508: Duplicate downlink in Multicast

Akior, 2020-07-17:
- [ x] The issue is present in the latest release.
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
Perform a POST  /api/multicast-groups/{multicast_queue_item.multicast_group_id}/queue 
app-server show 2 message on same BS with same fCnt
[image]
## What did you expect?
One message
## Steps to reproduce this issue

Steps:

1. add 2 NS, EU and RU (don't know this importance)
2. create multicast group
3. add device class C
4. generate api Post

body:
{
 "multicastQueueItem": {
  "data": "MTIz",
  "fCnt": 30000,
  "fPort": 5,
  "multicastGroupID": "29d20663-83b2-459c-9658-008ca6b7fdbb"
 }
}

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.10.0  |
| Network Server      | v3.9.0  |
| Gateway Bridge      | v3.9.1  |
| Chirpstack API      | REST API  |
| Geolocation         | Russia  |
| Concentratord       |         |

Akior, 2020-07-17:
Lora agent get exact one message, seems "cosmetic" bug

brocaar, 2020-10-01:
How many gateways do you have? In some cases it is expected that one downlink payload is sent multiple times, e.g. when all devices can't be reached by a single gateway.

Akior, 2020-10-09:
There are 6 gw.
But lora agent visible only from one.

I get screenshot from GW logs, and agent get only one message not two, so there are duplicate log bug ?
> How many gateways do you have? In some cases it is expected that one downlink payload is sent multiple times, e.g. when all devices can't be reached by a single gateway.

closed 2025-05-02 by brocaar