brocaar/chirpstack-application-server #513: Failed to connect to PostgreSQL server within docker-compose

sylveon, 2020-07-28:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
I ran `make build` and `make serve` within the `docker-compose` environment, but cannot run the server because it fails to connect to PostgreSQL, despite the postgres container running and having the right ports opened.

[image]
[image]

## What did you expect?
For the server to successfully connect to the database and start.

## Steps to reproduce this issue

Steps:

1. Run `docker-compose run --rm chirpstack-application-server bash`
2. Within the container, run `make dev-requirements ui-requirements`
3. Then, run `make serve`

## Could you share your log output?

```shell
WARN[0000] No configuration file found, using defaults. See: https://www.chirpstack.io/application-server/install/config/
INFO[0000] starting ChirpStack Application Server        docs="https://www.chirpstack.io/" version=3.11.0-3-gba94418
INFO[0000] storage: setting up storage package
INFO[0000] storage: setup metrics
INFO[0000] storage: setting up Redis client
INFO[0000] storage: connecting to PostgreSQL database
WARN[0000] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0002] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0004] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0006] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0008] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0010] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0012] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0014] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0016] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"
WARN[0018] storage: ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"

log repeated indefinitely...
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | 3.11.0-3-gba94418 |
| Network Server      | N/A        |
| Gateway Bridge      | N/A        |
| Chirpstack API      | N/A        |
| Geolocation         | N/A        |
| Concentratord       | N/A         |

brocaar, 2020-07-29:
The Docker Compose environment is only intended for running the tests, not run the ChirpStack Application Server as a service. To setup ChirpStack using Docker Compose please see: https://github.com/brocaar/chirpstack-docker/.

closed 2020-07-29 by brocaar

sylveon, 2020-07-29:
I want to be able to edit the application server code though, and the main docker container won't allow me to do that.

brocaar, 2020-07-30:
You can, but then you might need to extend the Docker Compose environment to not only configure a database for testing, but also for running the service itself. You then also might want to expose the ports :)

sylveon, 2020-08-01:
Managed to get it running in the end by modifying the chirpstack-docker compose file to expose postgresql and redis ports publicly as well as removing the chirpstack-application-server container. Copied the configuration file to a local build of this repo and it works after running `make serve`.

Yummybatsoup, 2021-12-19:
> Managed to get it running in the end by modifying the chirpstack-docker compose file to expose postgresql and redis ports publicly as well as removing the chirpstack-application-server container. Copied the configuration file to a local build of this repo and it works after running `make serve`.

@sylveon 
I got the same error but not able to solve it following your step, I would be grateful if you can list out the steps you have done to solve the problem. Thank You.