brocaar/chirpstack-application-server #519: OpenID Logout Not Implemented

jskillen, 2020-08-04:
- [X ] The issue is present in the latest release.
- [X ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
Created OpenID integration with Auth0. Login integration works. Logging out does not seem to clear the data from the session and redirect the user to the Auth0 logout endpoint. Consequently the user is not challenged with credentials after logging out.

## What did you expect?
When I return to the system after logging out I should be forced to log back in.

## Steps to reproduce this issue

Steps:

1. Implement the OpenID connector using Auth0.
2. Log in to Chirpstack and use Auth0 to log in.
3. Log out of Chirpstack.
4. Log back in. Observe that the Auth0 login immediately redirects to Chirpstack without challenging the user for credentials.

## Your Environment
ChirpStack Application Server 3.11.1

brocaar, 2020-08-05:
I think this depends on https://github.com/coreos/go-oidc/pull/226 to get merged in as this is the library I'm using for OIDC support.

brocaar, 2020-08-25:
@jskillen the above change makes it possible to configure a `logout_url` endpoint.

closed 2020-08-25 by brocaar

jskillen, 2020-08-25:
Thanks @brocaar !

commit 76608865 (2020-08-25): Add logout URL configuration for oidc.

When using for example Auth0, this makes it possible to configure the
Auth0 logout endpoint (potentially redirecting back to a login page).

This way, the user is not only logged out from the web-interface, but
also from the identity provider.

Closes #519.