brocaar/chirpstack-application-server #522: Random no acknowledge transmission error on confirmed uplinks

ReiniervdL, 2020-08-10:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
10% of confirmed uplinks do not receive an acknowledge, even after 5 x retries in Class A and C
Chirpstack gives an error message for each retry: 
type:"UPLINK_FCNT_RETRANSMISSION"
error:"frame-counter did not increment"

## What did you expect?
I would expect the system to send an acknowledge to the first confirmed uplink, or one of the retries.
The expected behavior was checked using the same set up connected to TTN.

## Steps to reproduce this issue

Steps:

1. send multiple confirmed uplinks (2 minutes separation)
2. 90% received as OK, 10% Tx-Nack
3.
4.

Log output:
Chirpstack console:

type:"UPLINK_FCNT_RETRANSMISSION"
error:"frame-counter did not increment"

Node diagnostic output on error:

13:56:01.634 -> Send data: 0000000000CE0000
13:56:01.634 -> 
13:56:02.924 -> send freq.904700000 DR0 TxPoweRX freq.925700000 DR10 symTimeout 144(4666407)
13:56:02.966 -> 
13:56:03.013 -> OnRadioRxTimeout 4666478
13:56:03.013 -> 
13:56:03.943 -> RX freq.923300000 DR8 symTimeout 72(4667414)
13:56:03.943 -> 
13:56:04.544 -> OnRadioRxTimeout 4668010
13:56:04.544 -> 
13:56:07.626 -> send freq.904300000 DR0 TxPower 0 counter 39(370688 us) 4671070
13:56:07.626 -> 
13:56:08.972 -> RX freq.924500000 DR10 symTimeout 144(4672422)
13:56:08.972 -> 
13:56:09.250 -> OnRadioRxTimeout 4672724
13:56:09.250 -> 
13:56:09.979 -> RX freq.923300000 DR8 symTimeout 72(4673430)
13:56:09.979 -> 
13:56:10.532 -> OnRadioRxTimeout 4674026
13:56:10.532 -> 
13:56:12.642 -> send freq.903900000 DR0 TxPower 0 counter 39(370688 us) 4676097
13:56:12.642 -> 
13:56:14.003 -> RX freq.923300000 DR10 symTimeout 144(4677450)
13:56:14.003 -> 
13:56:14.250 -> OnRadioRxTimeout 4677751
13:56:14.250 -> 
13:56:15.009 -> RX freq.923300000 DR8 symTimeout 72(4678457)
13:56:15.009 -> 
13:56:15.573 -> OnRadioRxTimeout 4679053
13:56:15.573 -> 
13:56:16.634 -> send freq.904900000 DR0 TxPower 0 counter 39(370688 us) 4680124
13:56:16.681 -> 
13:56:18.025 -> RX freq.926300000 DR10 symTimeout 144(4681476)
13:56:18.025 -> 
13:56:18.326 -> OnRadioRxTimeout 4681777
13:56:18.326 -> 
13:56:19.028 -> RX freq.923300000 DR8 symTimeout 72(4682484)
13:56:19.028 -> 
13:56:19.580 -> OnRadioRxTimeout 4683080
13:56:19.633 -> 
13:56:21.721 -> send freq.904700000 DR0 TxPower 0 counter 39(370688 us) 4685151
13:56:21.721 -> 
13:56:23.041 -> RX freq.925700000 DR10 symTimeout 144(4686504)
13:56:23.041 -> 
13:56:23.326 -> OnRadioRxTimeout 4686806
13:56:23.326 -> 
13:56:24.044 -> RX freq.923300000 DR8 symTimeout 72(4687511)
13:56:24.044 -> 
13:56:24.631 -> OnRadioRxTimeout 4688107
13:56:24.631 -> 
13:56:25.692 -> send freq.904100000 DR0 TxPower 0 counter 39(370688 us) 4689180
13:56:25.739 -> 
13:56:27.041 -> RX freq.923900000 DR10 symTimeout 144(4690531)
13:56:27.096 -> 
13:56:27.342 -> OnRadioRxTimeout 4690833
13:56:27.342 -> 
13:56:28.044 -> RX freq.923300000 DR8 symTimeout 72(4691539)
13:56:28.097 -> 
13:56:28.646 -> OnRadioRxTimeout 4692135
13:56:28.646 -> 
13:56:31.716 -> Tx_noACK

## Your Environment

  ChirpStack Concentratord:
   NA

  ChirpStack Network Server 3.11.0

  ChirpStack Gateway Bridge 3.9.2

  ChirpStack Geolocation Server:
  NA

[image]

[image]

brocaar, 2020-08-26:
TTN might allow for re-transmitting the ACK, I'm not sure. 

It is really hard to tell if this is a bug in ChirpStack, or something with your setup. E.g. it could be there are collisions in sending the downlink (e.g. the gateway is busy sending an other downlink). It could also be a timing issue at the device side causing it to not receive some downlinks.

I expect that ChirpStack did try to send back an ACK, but that it didn't make it to the device. It explains your situation as at the ChirpStack side, the uplink frame-counter was incremented. The device did not receive the ACK and performs a retry which is rejected by ChirpStack (and it should) as the frame-counter did not increment.

closed 2020-10-05 by brocaar

jacoscaz, 2021-07-26:
Hello! First of all, thank you for all the work that goes into ChirpStack. I've recently started experimenting with LoRa / LoRaWAN and it really is a complex protocol. Kudos to you for shipping such a wonderful project.

I think I am stumbling into something similar to this. I'm using an `Arduino MKRWAN 1300` that sends messages to a local gateway that forwards packages to the `ChirpStack Gateway Bridge -> Network Server -> Application Server` running as local Docker containers.

I am experiencing fairly predictable gaps due to `UPLINK_FCNT_RETRANSMISSION` / `frame-counter did not increment` errors. Transmitting around every 2 minutes I get a gap every 55 minutes or every 25 - 26 packets, as per the attached chart. I guess, and I might be wrong, that something is causing the loss of an ACK packet, after which the Arduino spends some time retrying sending the same packet again and again. However, this is happening with both the router and the Arduino board on my desk at a short distance between each other. The fact that these gaps are so reliably predictable even at these distances makes me think about a software issue.

Furthermore, I haven't been able to replicate this when connecting my gateway to TTN rather than ChirpStack while using the very same firmware on the Arduino board. For the time being I've disabled frame count checks via the Application Server's dashboard and added a couple of warning messages about duplicate frame counts in my application's backend. 

## Your Environment

ChirpStack Application Server: 3.15.0

ChirpStack Network Server: 3.13.0

ChirpStack Gateway Bridge: 3.11.0

ChirpStack Concentratord: NA

ChirpStack Geolocation Server: NA

[image]