brocaar/chirpstack-application-server #528: Error:”invalid MIC”

Shahboz0, 2020-09-03:
## What happened?
Hi,
I have problem with uplink, device joining by ABP.My frame counter is disabled,
screen of problem here
[image]
in gateway I saw packets
[image]
logs from network and application server show me no errors
[image]
my device-profiles parameters
[image]
My configs here
https://github.com/brocaar/chirpstack-network-server/issues/494

brocaar, 2020-09-07:
An invalid MIC means that the provided vs expected MIC do not match. Usually this means that your session-keys are invalid.r

closed 2020-09-07 by brocaar

Shahboz0, 2020-09-09:
But sometimes I'm getting uplink packet without error