brocaar/chirpstack-application-server #530: GET /api/service-profiles

noppsi, 2020-09-07:
- [ x] The issue is present in the latest release.
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
Hello

As soon as I use the ChirpStack Application Server REST API and use the URL "/api/service-profiles" to get a list of all service profiles I always get an error message. I tested it on two existing and one newly installed Chirpstack server and always got the same output. I also tried to give the Parameters "?limit=10&offset=0&organizationID=1" but it dident make any change.

Thank you

## What did you expect?

Somthing like this:

```json
{
  "result": [
    {
      "createdAt": "2020-09-06T18:23:42.694Z",
      "id": "string",
      "name": "string",
      "networkServerID": "string",
      "networkServerName": "string",
      "organizationID": "string",
      "updatedAt": "2020-09-06T18:23:42.694Z"
    }
  ],
  "totalCount": "string"
}
```

## Steps to reproduce this issue

Steps:

1. Call the API URL "http://{ip-adress}:8080/api/service-profiles" (and over the thes site http://{ip-adress}:8080/api#!/ServiceProfileService/List)
2. It answers the error

## Could you share your log output?

```json
{
  "error": "subject must be user",
  "code": 2,
  "message": "subject must be user",
  "details": []
}
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.12.1  |
| Network Server      |  v3.10.0       |
| Gateway Bridge      |  v3.9.2       |

jskillen, 2020-09-09:
Do let me know if you want me to open another ticket.

This also occurs when using the api to enumerate the network servers at the /api/network-servers endpoint when using an API key.

Safrone, 2020-09-23:
I also face this issue on listing devices /api/devices/

brocaar, 2021-02-02:
Sorry for the late follow-op @Safrone, but please create a separate issue for this :) This issue has been solved with #532.

closed 2021-02-02 by brocaar