brocaar/chirpstack-application-server #534: At least API call  /api/devices/{dev_eui}/keys  GET and POST doesn't work correctly

antoniohabland, 2020-09-19:
- [  X ] The issue is present in the latest release.
- [  X ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

I'm using the API call  /api/devices/{dev_eui}/keys  to programatically add nodes from a backoffice managment tool and everything is working properly EXCEPT the POST keys doesn't match the KEYS in the chirpstack-application-server.

Both network and app are version 3.12.1

when I make the POST  /api/devices/{dev_eui}/keys with this JSON 
{
  "deviceKeys": {
    "devEUI": "a84041e9d1821d68",                --> this is correctly set and get
    "nwkKey": "d6782bb1a6ed57bc9da54649bdeb256d",   --> this is inserted in the appKey  in the app-server !!!
    "appKey": "00000000000000000000000000000000",  --> this  parameter is not POST 
    "genAppKey": "00000000000000000000000000000000"  ---> neither
  }
}

so at the moment for to use it , I have to set nwkKey with the appKey in order to really set the appKey in the application-server !!!

on GET ==> the appKey and getAppKeys alway return 000 and the nwkKey contains the appKey 

Please check and continue with the good work !
Thanks

mmrein, 2020-09-21:
Dop you know about this https://www.chirpstack.io/application-server/changelog/#lorawan-11-support ?

antoniohabland, 2020-09-21:
Hi
didn't know that !  Thanks for the update.

closed 2020-09-21 by antoniohabland