brocaar/chirpstack-application-server #535: MAC commands logging(JoinReauest, JoinAccept, etc...)

golovlevviktor, 2020-09-28:
It will be great to have a responsibility to get information about MAC commands via MQTT interface - JoinAccepts, JoinRequests, StatusRequests, etc - It will give more information about network and it`s load.

For now it is possible to get this  information from live log, or parse protobuf messages from MQTT gateway/... topics

brocaar, 2020-10-01:
> It will give more information about network and it`s load.

Could you clarify this?

In general I'm not sure if I would favor this, but I'm open for discussing this further.

sagar-patel-sls, 2020-11-02:
@brocaar 
can we expose `LoRaAirtime` in uplink?

brocaar, 2020-11-02:
I'm closing this issue. @sagar-patel-sls yes that would be possible, but it would be for a different GitHub issue :)

closed 2020-11-02 by brocaar