brocaar/chirpstack-application-server #536: Kafka per application integration

oscni, 2020-09-28:
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

Enabling the possibility to setup sending payload data to kafka topic.

## What is the use-case?

Send payload data to kafka right now requires an intermediate transfer protocol such as HTTP. Sending it directly to kafka would be preferably.

## Implementation description

Pretty much setting it up the same way as the other per application integrations.

## Can you implement this by yourself and make a pull request?

brocaar, 2020-10-01:
Do you (or others) know what the overhead will be to connect, publish and disconnect for every event?
Per-application integrations will not maintain a connection state, while global integration do.

oscni, 2020-10-01:
No I and probably most other do not know how chirpstack works internally. Also in documentation it is not always so clear what is a global integration and what is an application specific integration so it makes it even harder to understand how the integrations should be used.

From a user perspective it seems problematic to have data from one application going to one kafka server and data from another application going to another kafka server. Chosing the global way means an intermediate system is probably needed in between to split up the message from chirpstack to the kafka servers. And per integration has the drawbacks you explain. Is there any plan for per application integration that maintains the connection state?

Any how, thanks for a great application so far :)

brocaar, 2020-10-02:
> Also in documentation it is not always so clear what is a global integration and what is an application specific integration so it makes it even harder to understand how the integrations should be used.

Contributions are always welcome! :) I do not deny that I have a bias, so things that are obvious to me might not be obvious to others. Contributions to improve the documentation are always welcome: https://github.com/brocaar/chirpstack-docs

> And per integration has the drawbacks you explain. Is there any plan for per application integration that maintains the connection state?

No, that would become problematic in terms of scalability, complexity and potentially also security. E.g. you might have 100 applications with each a Kafka integration which must maintain a connection. This number would multiply by the number of instances. There are probably smart ways to reduce the number of open connections, but then it would also become more complex.

oscni, 2020-12-23:
Per-application integrations will not maintain a connection state. A new issue to allow all per-applications to maintain connection state need to be fixed before this can be reopened.

closed 2020-12-23 by oscni