brocaar/chirpstack-application-server #537: Add gateway downlink information in TxAck message

hoffi085, 2020-09-29:
- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

The downlink information could be added to the TxAckEvent message. The downlink information could be:

- Gateway EUI which has emitted the downlink
- Tx information: frequency, spreadingfactor, bandwidth, code rate, tx power [dBm]
- Metadata of Gateway
- ...

## What is the use-case?
Since the network server chooses the best gateway for downlink in a given margin (Downlink gateway randomization since version 3.8.0) adding the gateway and transmitt information to the TxAckEvent could help to find problems when sending downlinks to the device.

## Implementation description
Adding the gateway information in the TxAckEvent is the best place for this feature as recommended by @brocaar [here](https://forum.chirpstack.io/t/getting-gateway-eui-which-has-send-the-downlink/9236)

sagar-patel-sls, 2021-01-06:
Hi, @brocaar 

I have implemented this feature and ready to make a pull request, but I would like to review the below screenshot of txinfo.

I have made changes in `chirpstack-api`, `chirpstack-network-server` and `chirpstack-application-server`

please find the below screenshot for your reference

**1. MQTT Integration TxAckEvent**
[image]

**2. Device Data TxAck** 
[image]

@brocaar please review this change
Thanks

sagar-patel-sls, 2021-01-08:
@brocaar any update?

sagar-patel-sls, 2021-02-03:
Hii @brocaar 
any update on this issue?

brocaar, 2021-02-08:
Sorry for the delay @sagar-patel-sls, I've had a really busy period past months supporting various projects. I'm now catching up with all the community requests and will now review your PR :)

PR #576 (2021-03-01): Add gateway downlink info in TxAck message
This exposes the tx_info field, holding all the TX parameters to TxAck Event

Fixes #537
closed 2021-03-01 by brocaar