brocaar/chirpstack-application-server #538: undeletable admin user

shepiloff, 2020-09-29:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

make default admin user undeletable

## What is the use-case?
in case of accidental of user deletion there are no users with admin priveleges and no users who can log in to UI

## Implementation description

## Can you implement this by yourself and make a pull request?

sagar-patel-sls, 2020-11-05:
Hi @brocaar 

we cannot allow delete users themselves.

If you agree then I will make PR for this
Thanks

brocaar, 2020-11-09:
Hi @sagar-patel-sls, I think making sure that users can't delete themselves would be better than making the default user undeletable :) I'm happy to review a PR for that!

PR #557 (2020-11-12): Users can't delete themselves from the user
Closes #538
closed 2020-11-12 by brocaar