brocaar/chirpstack-application-server #539: Register device keys, encryption keys does not have byte order

connyan, 2020-09-29:
## What happened?
When registering a device's encryption keys there is a button called MSB which can be used to toggle byte order between MSB and LSB. This can be confusing for some users and is also wrong, encryption keys are just a sequence of bytes.

## What did you expect?
The button should be labeled "reverse" or similar. Comment on button hover should be to check against device specs.

## Steps to reproduce this issue

Steps:

1. Register a new device
2. Go to keys tab
3. Click the MSB button, the bytes reverse and the button label changes to LSB
4. Hover over button, a comment about LSB appears

## Your Environment
chirpstack-application-server:3

shepiloff, 2020-09-29:
LSB/MSB  is ok 
all endnode device manufacturers marks device keys as LSB/MSB sequence.  that's standard.

brocaar, 2021-03-31:
I'm closing this issue as I think LSB / MSB is the correct way.

closed 2021-03-31 by brocaar