brocaar/chirpstack-application-server #540: Can't change admin password (because no email address)

lafrech, 2020-10-01:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

Fresh install. Logged in as admin:admin. As stated in the docs, first thing, tried to change password. Couldn't. Got the dreaded "invalid e-mail (code: 3)" error as in #287.

It doesn't really make sense to me to add an email address to a virtual account such as admin, so I created myself an account with my own email address. Then I tried to disabled admin. Couldn't either because no address. Ultimately, I had to enter an email address to admin as well. Then I could disable it.

Perhaps #287 should be reconsidered.

## What did you expect?

Be able to change admin password seamlessly as first action.

## Steps to reproduce this issue

Steps:

1. Install.
2. Login as admin:admin
3. Change password

## Your Environment

Using Chirpstack Application Server version 3.12.2 (debian package)

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.12.2  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2020-10-01:
I will ignore the email validation for `admin`. That should solve the issue I believe.

closed 2020-10-01 by brocaar

lafrech, 2020-10-01:
That was fast!

:+1:

commit d7ac9899 (2020-10-01): Exclude email validation for admin user.

Closes #540.