brocaar/chirpstack-application-server #547: Chirpstack Server Downlink Frequency Bug

lehungmt94, 2020-10-15:
- [x ] The issue is present in the latest release.
- [x ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?
Chirpstack Server Downlink send incorrect frequency to Gateway Lora
This is my [log file](https://github.com/lehungmt94/STM8S003-Sample-Project/blob/master/gateway-da1ea43466778888.json)

Downlink with ACK by frequency  433mHz but Downlink send data by user is frequency 0mHZ
This is my post in forum Chirpstack
[https://forum.chirpstack.io/t/have-problem-with-frequency-downlink/9463](https://forum.chirpstack.io/t/have-problem-with-frequency-downlink/9463)
## What did you expect?
I expect Chirpstack Server send correct frequency I setting on Gateway-profiles

shepiloff, 2020-10-15:
is the  LoRaWAN regional band selected correctly in your chirpstack-network-server.toml  ?

lehungmt94, 2020-10-16:
Thank you @shepiloff  
This is my  chirpstack-network-server.toml

```
[postgresql]
dsn="postgres://chirpstack_ns:chirpstack_ns@postgresql/chirpstack_ns?sslmode=disable"

[redis]
url="redis://redis:6379"

[network_server]
net_id="000000"

[network_server.band]
name="EU433"

[network_server.network_settings]

  [[network_server.network_settings.extra_channels]]
  frequency=433375000
  min_dr=0
  max_dr=12

  [[network_server.network_settings.extra_channels]]
  frequency=433175000
  min_dr=0
  max_dr=12

  enabled_uplink_channels=[0,1,2,3]

[network_server.gateway.backend.mqtt]
server="tcp://mosquitto:1883"

[join_server.default]
server="http://chirpstack-application-server:8003"

[geolocation_server]
server="chirpstack-geolocation-server:8005"
```

brocaar, 2020-10-16:
@lehungmt94 is this related to an ABP device?

Then please check that you have filled in all relevant device-profile fields. A common mistake is to leave the RX2 freq. to `0`, resulting in the above issue. After updating the device-profile, make sure to re-activate the device.

closed 2021-03-31 by brocaar