brocaar/chirpstack-application-server #551: pq: column gp.stats_interval does not exist

civilizeddev, 2020-10-19:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

- An error message `pq: column gp.stats_interval does not exist` popped up when loading Dashboard.
- Both Dashboard and Org. Dashboard
- With v3.12.1 it worked.

[image]

## What did you expect?

- No Error Message

## Steps to reproduce this issue

Steps:

1. Upgrade the docker image version `chirpstack-application-server:3.12.1` to `chirpstack-application-server` (I'm not sure if it correctly refer to 3.12.2)
2. Connect the Chirpstack Application Console.
3. Click the Dashboard.
4. (Error)

## Could you share your log output?

```shell
time="2020-10-19T05:23:47Z" level=error msg="finished unary call with code Unknown" ctx_id=e8ae2b3f-d2ec-4d48-82e7-fe39216f3ddd error="rpc error: code = Unknown desc = pq: column gp.stats_interval does not exist" grpc.code=Unknown grpc.method=GetGatewaysSummary grpc.service=api.InternalService grpc.start_time="2020-10-19T05:23:47Z" grpc.time_ms=5.847 peer.address="[::1]:60984" span.kind=server system=grp
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.12.2  |
| Network Server      | v.3.10.0 |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

chopmann, 2020-10-19:
did you migrate the tables?
https://github.com/brocaar/chirpstack-application-server/blob/master/cmd/chirpstack-application-server/cmd/root.go#L43
automigrate should be true

civilizeddev, 2020-10-20:
@chopmann You're right. Thank you.

closed 2020-10-20 by civilizeddev