brocaar/chirpstack-application-server #558: ADR engine - NbRep : Please can we have the option to control Packet Repeat Rate  in the Service or Device Profiles

Cloolalang, 2020-11-18:
**

- [ x] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## Summary

In Service profiles we can modify the ADR engine slightly with MAX/MIN data rate, and in Device profile we have MAX device EIRP.
Both if these seem to work with the ADR engine process.
It would also be good to be able to control  NbRep  as different networks experience various levels for packet-collision threat and packet loss of other reasons. According to the Semtech algorithm, the target is about 10%.
Anyway to be able to turn on/off this part of the engine would be very useful for testing and optimizing situations.
 

## What is the use-case?
Lab- test networks
Networks which can accept various levels of packet loss.

## Implementation description

In either the device or service profile UI page.

## Can you implement this by yourself and make a pull request?
Sorry not the skills

Cloolalang, 2020-11-18:
Or as an alternative:
In the chirpstack-network-server.toml
a field:

disable NbRep = true

brocaar, 2021-06-15:
I'm closing this issue. For lab testing, I suggest to use the ADR plugins which was recently introduced. I don't think it is a good idea to give all end-users the control over the used DR and TX Power as this might cause issues on multi-tentant deployments.

closed 2021-06-15 by brocaar