brocaar/chirpstack-application-server #559: Copy key functionality doesn't work: Application Server v3.13.0

bryanstein, 2020-11-19:
- [X ] The issue is present in the latest release.
- [X ] I have searched the [issues](https://github.com/brocaar/chirpstack-application-server/issues) of this repository and believe that this is not a duplicate.

## What happened?

Attempts to copy the key generated by chirp stack application server fail when using the provided copy icon. I observed this in the last release also. I'd like be able to choose between copying a hex string or hex array and this functionality doesn't work either.

## What did you expect?

I expected to be able to copy the generated key by clicking the copy icon; additionally I expected to be able to copy the data as either a hex array or hex string by selecting either of the two options. 

## Steps to reproduce this issue

Steps:

1. Create a device
2. Click on the device 
3. Navigate to Keys(OTAA)
4. Click Generate Random Key
5. Click the two vertical rectangles to copy the key
6. Attempt to select Hex String or Hex Array and it will fail.
## Could you share your log output?

Application Server  v3.13.0

Urenthummar, 2020-12-08:
Also these things not work on past release. @brocaar  can u solve this tiny issue.

mmrein, 2020-12-09:
May be browse issue somehow. I can confirm this on AS v3.12 and latest version build from source with Firefox on W10 and Ubuntu 20.

It seems to be working with Chromium on Ubuntu 20 for example, others not tested.

mcrobs, 2021-03-05:
I have a similar error, this is the output on the chrome console
~~~
react-dom.production.min.js:52 Uncaught TypeError: Cannot read property 'writeText' of undefined
    at onClick (AESKeyField.js:42)
    at Object.s (react-dom.production.min.js:14)
    at d (react-dom.production.min.js:14)
    at react-dom.production.min.js:14
    at g (react-dom.production.min.js:15)
    at at (react-dom.production.min.js:52)
    at it (react-dom.production.min.js:51)
    at st (react-dom.production.min.js:52)
    at dt (react-dom.production.min.js:56)
    at N (react-dom.production.min.js:287)
react-dom.production.min.js:52 Uncaught TypeError: Cannot read property 'writeText' of undefined
    at onClick (AESKeyField.js:34)
    at Object.s (react-dom.production.min.js:14)
    at d (react-dom.production.min.js:14)
    at react-dom.production.min.js:14
    at g (react-dom.production.min.js:15)
    at at (react-dom.production.min.js:52)
    at it (react-dom.production.min.js:51)
    at st (react-dom.production.min.js:52)
    at dt (react-dom.production.min.js:56)
    at N (react-dom.production.min.js:287)
~~~

PR #593 (2021-05-10): Fix Copy key functionality
Fixes #559
closed 2021-05-10 by brocaar